tock/tock #1: rustc expects libgcc to have __aeabi_memcpy

phil-levis, 2015-06-25:
Not sure if this is due to my gcc installation being different, or a common problem.

phoenix:tock pal$ git log | head -1
commit 61d90bbc57687e77ed5d0405a010f4ff41d22ef4
phoenix:tock pal$ make
Linking build/main.elf
arm-none-eabi-gcc -g -O3 -std=gnu99 -mcpu=cortex-m4 -mthumb -nostdlib -Tsrc/chips/sam4l/loader.ld build/crt1.o build/arch.o build/main.o -o build/main.elf -ffreestanding -lgcc -lc
build/main.o: In function `main':
main.0.rs:(.text.main+0x426): undefined reference to `__aeabi_memcpy'
phoenix:tock pal$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GCC) 4.8.3
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

phoenix:tock pal$ 
collect2: error: ld returned 1 exit status
make: *** [build/main.elf] Error 1

phil-levis, 2015-06-25:
Included implementation in src/chips/sam4l/crt1.c, simply calls memcpy().

closed 2015-06-25 by phil-levis

alevy, 2015-06-25:
This is almost certainly an arm-none-eabi-gcc version issue. The version I'm using is 5.1.0. The argument order is often different with __aeabi_* prefixed functions, so it's much better to rely on the compiler's libc if possible. I think there were other bugs in the version of arm-none-eabi-gcc on OS X, so it's worth upgrading anyway.

ppannuto, 2015-06-25:
The problem is in part `-nostdlib` which removes all standard library functions from the compilation (for good reason).

The gcc specs, however, say that compiler is permitted to emit calls to a few well-defined functions (memcmp, memcpy, memset, and memmove I believe are the whole set) as part of it's optimization efforts. The canonical way to address this is to link in `libgcc.a` (add `-lgcc` to your compilation) which includes platform-optimized implementations of all of the needed methods.

The gcc man page section on `nostdlib` goes into this in a little more depth iirc

phil-levis, 2015-06-25:
Pat, the issue is __aeabi_memcpy, for which there is no implementation. I implemented it to call memcpy (for which there is an implementation from libgcc.a).