tock/tock #18: Buffers should be adjusted for different boards

lizardo, 2016-01-15:
Firestorm has 64K RAM , while nRF51822 based boards have "only" 16K. Tock is able to run fine on smaller systems, as long as we are able to scale down static buffers or limit memory possible using modular designs.

One example is per app memory reservation. Current master uses a hard-coded 8K limit, but that seems too much for 16K systems.

0xc0170, 2016-01-23:
Can you provide more details, the reference where is the limit hard-coded?

alevy, 2016-01-24:
@0xc0170 The hard coding is in `src/process/process.rb` in the static variables `PROC_MEMORY_SIZE` and `NUM_PROCS`. Ideally this should be platform or chip specific, but the trivial way of doing that would create a circular dependency between those crates.

alevy, 2016-02-05:
Duplicate of #30

closed 2016-02-05 by alevy