tock/tock #22: Generating libcore for architecture

0xc0170, 2016-01-23:
Makefile for libcore is currently broken if you build for more than one platform There are two platforms, let's just say one is cortex-m0 and cortex-m4 (different thumb versions). Triple does not match for core lib. The error output:

```
./tock/src/process/lib.rs:1:1: 1:1 error: couldn't find crate `core` with expected target triple ./tock/src/chips/nrf51822/target.json [E0461]
./tock/src/process/lib.rs:1 #![crate_name = "process"]
                                              ^
./tock/src/process/lib.rs:1:1: 1:1 note: crate `core`, path #1, triple ./tock/src/chips/sam4l/target.json: \\?\C:\Code\github\rust\tock\build\libcore.rlib
error: aborting due to previous error
make: *** [build/libprocess.rlib] Error 101
```

This is caused by copying libcore to build directory without appending any triple specific value. A quick fix would be to copy libcore to a platform specific build directory. A better one might be to come up with a schema from target.json file. What shall we consider, arch, llvm-target?

alevy, 2016-01-25:
This is actually not limited to libcore. Each rust crate is compiled for a specific target, so, in fact, all the drivers and common kernel components need to be recompiled as well. Simiarly, applications may need to be compiled with different GCC targets, so they too need to be recompiled.

Ideally the build directory could maximize artifact sharing by splitting have a separate build directory per chip and platform (so multiple platforms using the same chip could reuse a compiled libcore and lib-_chipname_.

But a separate build directory for each platform would be a good start.

alevy, 2016-02-05:
Duplicate of #30

closed 2016-02-05 by alevy