tock/tock #23: Option to have apps in an external repo?

bradjc, 2016-01-26:
I imagine that the plan long-term is to not have the `apps` folder in with the kernel, and I think it makes sense for it to stay there for quick testing now, but what would it take to have a separate repo with apps that we can upload to a firestorm that already has the kernel?

I ask because my ble-serialization app will need a bunch of nordic sources in the short-term and I don't want to complicate the kernel build with the external dependency.

alevy, 2016-01-26:
RE: long run, yes. Might keep apps as a kind of `examples` folder, but otherwise, yeah.

I don't think it would take that much to allow creating a `main.elf` to program onto the board with an app taken from somewhere external. I think it would just be a matter of adding a make target that looks incorporates the ELF specified in some environment variable as apps in `main.elf`.

Depending on how far @ppannuto gets making the build system better, this might be pretty trivial.

alevy, 2016-02-05:
Duplicate of #30

closed 2016-02-05 by alevy