tock/tock #30: Revamp build system

alevy, 2016-02-05:
The current build system is pretty cobbled together and is pretty hard to understand (and therefore change). We would like the system to allow at least:

  * [ ] More modular build system (modeled after ??? )
  * [ ] Ability to build app in external repo
  * [ ] Generate articfacts for multiple boards/apps/chips without having to do a clean-all
  * [ ] Per-board or per-chip variables that show up in different crates (e.g. memory sizes for apps).

alevy, 2016-02-28:
@ppannuto merge a partial revamp of the build system which does a better job of tracking changes in apps and refactors most of the common build code from apps out. The result is that less code needs to be copied into a new app and it should be much easier to build an app outside of the kernel source directory.

Building an app + kernel is pretty different though and we should update the README to reflect that.

I believe @ppannuto is concurrently working on moving the kernel's build system to cargo.

alevy, 2016-02-28:
@ppanuto, as of #45 I'm encoutering three issues with the build system:

  * [ ] The kernel doesn't built totally from scratch (without a `build` directory) from the root directory. I think this is because of a hold-over depedency in `all` on `BUILD_PLATFORM_DIR`.

  * [ ] Nothing builds from scratch from an app directory because of the following error: `No rule to make target '../../build/storm/ctx_switch.o', needed by '../../build/storm/c_blinky/kernel_and_app.elf'.  Stop.`

  * [ ] `make clean` tries to `rm -Rf /*.*` which is obviously bad. Again I think this is a hold over from `BUILD_PLATFORM_DIR` being defined.

  * [ ] `make clean-all` only cleans the current platform's build directory (not sure if this is necessarily bad but is a bit surprising)

ppannuto, 2016-02-28:
@brghena fixed the rm /*.* yesterday in 6769c97, I'll look into the rest.

ppannuto, 2016-02-28:
@alevy I can't repro not building from scratch? I tried both `rm -rf build` and a clean checkout of tock. I did remove the spurious BUILD_PLATFORM_DIR dependency in 9941ed6, but it has worked correctly from me before that as well.

ppannuto, 2016-02-28:
As of 643ee19 these should all be fixed.

alevy, 2016-02-28:
@ppannuto Yes confirmed!

closed 2016-07-11 by alevy