tock/tock #41: Investigate need for built-in function replacements in crt1.c

lizardo, 2016-02-10:
There is a TODO on the crt1.c but it is easier to track here.

Currently, we implement replacements for several built-in functions that could be provided by some runtime library (GCC or LLVM). It is not clear (at least for me), why they are needed. Are we missing linking to some library that could provide those functions?

alevy, 2016-02-28:
I removed the built in functions in cf0de8fe934de5b92223ad48353f44102bae1fcd

I believe this is resolved except I've heard one report of `memclr` and `memclr4` not being available at kernel link-time. I'm guessing this has to do with the version of newlibc perhaps. but still investigating

alevy, 2016-04-05:
Indeed it had to do with the version of newlibc. Since no one else has complained so far, I'm marking this issue closed.

closed 2016-04-05 by alevy