tock/tock #42: Use Jinja2 templates for code generation from CMSIS SVD models

lizardo, 2016-02-10:
From the original [discussion](https://github.com/helena-project/tock/pull/38#issuecomment-182235712):

"Jinja2 templates are a good way to decouple the model from the "view" (i.e. generated Rust/C code). In fact, that reminded me Yasha (https://github.com/kblomqvist/yasha) already supports SVD models through CMSIS-SVD! There is one example for nRF51 that generates zinc ioreg! macros: https://github.com/kblomqvist/nrf51.rs/blob/master/nrf51.rs.jinja."

lizardo, 2016-03-08:
Turns out the CMSIS-SVD support in yasha is not fully compatible with how we are using the model (for instance, it seems to assume the .svd is bundled together with the template). For now, I'm using Jinja2 + cmsis-svd module instead.

Already converted peripheral_registers.rs. Still missing ISR. Next will be NVIC indexes.

alevy, 2016-08-21:
I think this is resolved. @lizardo feel free to re-open if not.

closed 2016-08-21 by alevy