tock/tock #43: Create read-only/write-only/read-write variants of VolatileCell

lizardo, 2016-02-10:
Currently VolatileCell assumes every register is read-write. Ideally, we should provide RO/RW/WO variants that enforce correct access permissions at development time.

Proposed by @0xc0170 on https://github.com/helena-project/tock/pull/38#issuecomment-182235712

alevy, 2016-08-21:
Here's a sketch of how that might look:
https://play.rust-lang.org/?gist=c72fff775926f381bf6371c2de24e995&version=stable&backtrace=0

alevy, 2016-08-21:
Would also be useful to have a handful of common, more specific types for as type parameters to `VolatileCell` for registers. For example, it would be nice to have types that:

  * Take up 32 bits in memory but only have 16/8 bits meaningful
  * Expose a bit setting/clearing interface
  * Others?

ppannuto, 2016-08-21:
Bit set, clear, and toggle (xor) are the standard triple for register
interfaces

alevy, 2016-08-22:
Agreed. Plus it's fairly common to have a register that controls both set & clear, so you end up doing something like:

```rust
registers.my_reg.set(registers.my_reg.get() & !(1 << BITNAME));
```

brghena, 2018-02-28:
Essentially solved by #658

closed 2018-02-28 by brghena