tock/tock #46: Watch for #[naked] support to come into Rust

ppannuto, 2016-02-28:
Our interrupt handlers are currently emitting function prologues that they should not, e.g:
```
00016b9e <reset_handler::h0a16a35074fed14diQc>:
   16b9e:       b580            push    {r7, lr}
   16ba0:       466f            mov     r7, sp
```

The way to fix this is to mark the functions as `naked`, but that hasn't quite made it into rust yet: https://github.com/rust-lang/rfcs/pull/1201

We should monitor this though and add it to the list of things worth updating rust versions for whenever it lands.

alevy, 2016-02-28:
I'm actually not sure they can be totally naked. For example, while the harware stacks the `LR` register, it does not stakc the `r7` (and other non-argument+general purpose registers).

ppannuto, 2016-02-28:
Right, not every interrupt handler should be naked, but some should, Reset, HardFault, and some of the other unrecoverable ones.

Relatedly, the ISR array should actually be a mixture of `extern fn()` and `extern fn() -> !` functions. Currently the compiler emits an unneeded `bx lr` to the end functions that it shouldn't but I don't know how to define an array of mixed type (if we even can). [I'm working on a replacement HardFault handler right now that panic!'s with some context for where the fault happened]

alevy, 2016-07-11:
It's been in nightly since March now: rust-lang/rust/pull/32410

So now a prerequisite is to upgrade our nightly version.

alevy, 2016-08-21:
This is resolved by #75 and #90

closed 2016-08-21 by alevy