tock/tock #51: App loading error with global `char * = "..."`

alevy, 2016-02-29:
The current application loading mechanism relies on two different mechanism for dealing with relocated text and data segments. Unfortunately, the method for dealing with the text segment doesn't capture all cases. In particular, there are ways to get a GOT entry which is a double pointer to the text segment, in which case, the Tock app loader doesn't correctly rewrite the address to the app's actual flash location.

@brghena Wrote a minimal example of this in the `app_load_error` branch. Basically, a global pointer to a static variable will create such a problem:

```c
char *foo = "hello";
```

One solution is to just locate even static variables in the data segement -- which is a simple one line change in the linker script. The main downside seems like it might be a waste of RAM. For example, all strings would be located in RAM as opposed to code.

What else could we do?

brghena, 2016-03-01:
Unfortunately mapping static variables to the data segment does not fix the problem.

The error here is that a variable is being set to a pointer at compile-time and does not have opportunity to be fixed up. The value of `foo` lives in the data segment, which we do not edit.

One terrible hack which would at least warn us when this is occurring would be to write a script that parses the data section in the lst for values matching 0x8000XXXX. Most likely they are invalid pointers...

The question @alevy raised in chat is how statically compiled libraries handle this problem. We should look into this more?

ppannuto, 2016-03-01:
So, I started looking into this a little. As a stopgap, 0840f36e will detect when this happens and throw an error.

The insight from that check is that the elf file does contain enough information to know that there is a symbol in rodata that needs to be relocated. If that symbol is marked for absolute relocation, then we'll hit this problem. It's possible that some combination of `-shared` or `-pie` could convince the tools to emit different relocations. Failing that, it's also possible that we could parse the elf ourselves and make a list of these. The actual relocations are relatively easy for our loader to patch up, it just need to know that it needs to do it.

ppannuto, 2016-03-01:
I mentioned this ARM tech doc on the call http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dai0242a/BCGHDBCB.html

It covers how these relocations are handled by a dynamic linker, which is related to, but not exactly what we want to do

ppannuto, 2016-03-01:
I should add that this was linked to from https://sourceware.org/bugzilla/show_bug.cgi?id=16163 , with this confidence-inspiring quote

> Of course the next question is "does this mean that the linker is generating illegal shared objects", at least as according to the AAEABI ?  I am afraid that the answer appears to be "yes", and I do not have a solution to this.  I think that it may be that established practice is that loaders need to be able to handle these relocations, even though strictly speaking they should not.

Again, not exactly our issue, but close enough that some material here may be useful.

alevy, 2016-03-01:
It's worthwhile asking what "good" output would be.

### What happens now

At the moment, this is how GCC compiles a pointer dereference of using the GOT, assuming register `r9` holds the GOT's base address:

If we have a `char const*` or a `char[]` we get...

```
  ldr r3, [pc, #28]  ; Load offset from GOT into r3
  ldr r3, [r9, r3]   ; Load the offset from the GOT into r3
```
Which works.

Conversely, if we have a `char *` or a `const char*` we get one more level of indirection, which kills us

```
  ldr r3, [pc, #28]  ; Load offset from GOT into r3
  ldr r3, [r9, r3]   ; Load the offset from the GOT into r3
  ldr r3, [r3, #0]   ; Load the memory referenced in r3
```
Where that second to last value of `r3` contains some compile-time determined, fixed address, which is an incorrect address.

This difference between the implementations is important because `char*` or `const char*`, while referencing static data, is itself a global variable that is allowed to change.

### What would we want

It seems like there are two options for how correct output would look like.

First, the loader in the kernel could (somehow) figure out where to patch the symbols, so the memory referenced after indirecting from the GOT actually contains a runtime-valid address. This would require knowing about the symbol table at load time. Is there a way to compile that into binary? This seems the more reasonable option.

Second, the compiler could add another level of indirection through the GOT. The memory referenced after indirecting from the GOT (the first time) would actually contain another offset into the GOT, which would contain the actual address. Of course, the pointer could be change to point to non-static data, so there would have to be a way for the caller to differentiate between GOT referenced data and dynamically allocated data (which is obviously not present in the GOT). The code for caller is probably too complex.

alevy, 2016-03-03:
We believe we've managed to fix the problem in f7c7243.

The linker is able to output sections of the form `.data.rel.*` for portions of the data segment that require relocation. By placing those sections together in the resulting binary and wrapping them in linker exposed variables, we can extend the `load_info` structure at the beginning of the app binary with offsets to the relocated data section.

The layout of the relocated data section is the same as the GOT, so in the kernel's app loader we can just reuse the same code to update those pointers at app load time.

Waiting on a bit more confirmation (probably from trying it on the BLE code @bradjc has been working on), and then will merge this into master.