tock/tock #53: Non-reentrant problem with putstr (and also all other synchronous I/O)

alevy, 2016-04-05:
With the current implementation of `wait` and `wait_for`, an app that mixes synchronous and asynchronous calls can easily run into trouble.

The high-level problem is that a call to a synchronous function (such as `putstr`) might be interleaved with a callback for that calls the same function. Of course, if that function is non-reentrant, this is a problem, but it's actually not possible to make a function reentrant while using `wait_for` "correctly".

There is a solution to this. the tl;dr is to change `wait_for` to wait for a boolean flag to flip to true rather than using a unique value of the return pointer. But first some background and an example of the problem before a detailed proposal for the solution.

### Background

First, a review of how `wait` and `wait_for` currently work. If you feel like you have a good grasp on this, you can skip down.

There are three related calls:

  1. `int __wait(void)` is a system call that blocks execution of the app until a new event is ready. At that point, the kernel signals the event by pushing a frame onto the app's stack and jumping to a callback function specified in a corresponding `subscribe` syscall.  When the callback returns, execution is resumed at the point where `__wait` was called, and the return value of the callback becomes the return for the `__wait` call.

  2. `wait_for(int)` is a library wrapper around `__wait` that keeps yielding until a callback of a particular type (specified by the argument) has been invoked. Any event that doesn't match is put onto a queue so that other calls to `wait_for` or `wait` (see \#3) see it (in case it corresponds to the event _they_ are waiting for). This is meant to correspond to a particluar event, such as `putstr` completion, has occured. To do this, all callbacks have to agree to use their return value to signal which event they responded to. To manage this, we currently have a global enum containing each event type in `apps/lib/firestorm.h`.

  3. `wait` is a library wrapper around `__wait` that first checks queue populated by `wait_for` and only actually calls the system call if the queue was empty.

### Why can't we make `putstr` reentrant

From a library writer's perspective, making `putstr` reentrant should be straight forward. I have a data sink (e.g. the console driver) that can only knows how to deal with one outstanding write request at a time, so I just need to buffer concurrent calls to `putstr`. The callback `putstr` uses to get notified that the console has finished doing it's thing will process the next buffered string to print _et voila_.

The problem is that this doesn't let us differentiate in `wait_for` between different calls to `putstr`. This means that when calls to the same synchronous function are interleaved, we can't be sure that the action has completed when we return. This maybe isn't so terrible with `putstr`, but this can definitely be a problem with order dependent operations. For example:

```c
timer_subscribe(my_callback, NULL);
timer_oneshot(1); // a very short timer
spi_write_sync("blahblah", 8);

CB_TYPE my_callback(...) {
  // Here, we're not sure if the following command will be buffered or not,
  // because "blahblah" may not have completed yet.
  spi_write_sync("cmd", 3);
  // Maybe "cmd" completed but only maybe "blahblah" completed...

  some_call_that_should_only_be_invoked_after_cmd_has_completed();

  return TIMER;
}
```

As a hack explore what was possible for `putstr`, I generated a unique return value for each call (just by incrementing a global counter), and waiting on that value instead of `PUTSTR` in wait for. With some slight modifications to `wait_for` (have to check if the callback is in the queue aftrer each call to `__wait`). That functionally works, but is a problem since return values to callbacks should be unique, and it's not clear how to make sure they don't overlap 

### Proposed Solution

The core problem is waiting on a something unique for each invocation of a synchronous call. It's hard to generate globally unique values, but it's not hard to generate globally unique pointers.

Instead of waiting on a return value in `wait_for`, wait on a particular boolean variable flipping to true. The callback doesn't have to return anything particular, instead it flips the boolean. The boolean can be a global (in the case of a non-reentrant function or if we're only ever processing the head of a globally stored list, e.g.) or can be passed in as the user-data parameter to the callback.

#### Benefits

The implementations for `wait` and `wait_for` are much much simpler (almost not worth implementing):

```c
void wait() {
  __wait();
}

void wait_for(bool* cond) {
  while(!*cond) {
    __wait();
  }
}
```

Writing callbacks that don't correspond to a `wait_for` is totally straight forward since you don't have worry about which value you are returning.

#### Disadvantages

These implementations discard the return value completely, even though in principal they could be repurposed (for example to return the length of the written to the console), but that would be a pretty complex implementation and that can always be simulated, again, by modifying another shared variable passed through the callback's user-data.

It requires changes to all functions that currently use `wait_for`. So far, i've ported most of them relatively easily. Porting them illuminated that many functions are non-reentrant, and maybe some should be.

Finally, in some cases (like Phil's rf233 code), where synchronous calls are not explicitly broken out, it becomes subtle where to set global boolean flags to false before waiting (it needs to be _before the first `wait` that might result in a callback, not just the corresponding `wait_for`_).

alevy, 2016-04-05:
/cc @phil-levis in particular

phil-levis, 2016-04-06:
Moving to a condition variable/signal approach makes sense. Definitely better. But also no locks needed because of our concurrency model.

I agree that it’s a bad idea to make all non-reentrant calls enqueue and return immediately, because it breaks ordering. But for some calls it’s OK. E.g., normal printf() does this, it doesn’t actually write() until a buffer is full or there is a newline. It ensures that all calls to printf will appear in the order they’re issued, but doesn’t promise that they have sequential consistency with the rest of the code.

Phil

alevy, 2016-04-06:
OK, so I believe I've finished porting everything in userland to use this (in a branch), and it seems to work fine.

Feedback sounds like it's reasonable to merge this?

I'll look into making printf return immediately in some cases.

closed 2016-04-08 by alevy