tock/tock #73: RTC on nRF51822

phil-levis, 2016-07-20:
There are currently 2 RTC implementations in the nRF; rtc.rs uses RTC1 for an underlying timer and systick.rs (in the nrf_pc10001) uses RTC0 for systick. We should integrate these two so the same RTC, RTC1, is used both for systick as well as the underlying timer clock. We need to use RTC0 for the BLE stack. Since they share a single NVIC entry this means a capsule that exposes all 3 and demultiplexes their events in software.

phil-levis, 2016-07-21:
It looks like the radio has to use RTC0; there are statically reserved PPI channels that tie radio events/tasks to timer0 (Table 90 on page 74 of v3.0 of the reference manual).

alevy, 2016-07-22:
This should be fairly straight forward with `virtual_timer` in `drivers`, but using that relies on (once again) flipping the dependency tree so we can pass systick into the scheduler rather than statically bind it.

closed 2016-10-25 by alevy