tock/tock #101: What memory manager are you using?

vitiral, 2016-08-25:
I have written a ultra lightweight memory manager for microcontrollers that completely avoids issues with fragmentation 

https://github.com/vitiral/tinymem

It's currently written in C and has a known bug, but I've been itching to rewrite it in rust. Would it be useful to you? 

To get it to work transparently, I think it would require a compiler plugin - as it uses indirection of pointers to allow it to move memory around safely. But from what I know about rust, it should be possible to implement.

vitiral, 2016-08-25:
I might be able to implement it by just using something like Box. This way, all heap allocations would have to go through Box, and also it would allow for allocations to fail (necessary for uC!)

alevy, 2016-08-25:
In general, we don't use a memory manager -- the kernel (on purpose) does not use dynamic memory allocation _at all_. Processes are free to use whatever they want, and our current runtime just uses `newlibc`'s malloc.

The one exception to this is that kernel drivers can "borrow" memory from a special region in the process called a "grant" region (currently `Container` in the code, but we'll change the name eventually). For that we currently don't have a proper memory manager yet, so that might be a good place to contribute if you're interested.

I'm  closing this issue, just because it's probably more appropriate to discuss this on the (recently minted) [mailing list](https://groups.google.com/forum/#!forum/tock-dev)

closed 2016-08-25 by alevy

vitiral, 2016-08-25:
Ok, good to know that you don't have anything yet. I will take a look and see how easy it would be to implement this in Rust (fairly simple I think!)

phil-levis, 2016-08-25:
@vitiral - it's actually a design point of the kernel that it doesn't have a memory allocator. Generally speaking, I've found that global memory allocators in embedded systems are a bad idea, because they couple failures between all parts of the system. I.e., if one piece of buggy code exhausts the heap, everything else can fail. Instead, in cases where you do want a form a dynamic memory allocation, it's better to create local, private allocators from fixed blocks of static memory (e.g., like typed slab allocators). E.g., a routing layer might allocate a pool of 12 packets.

vitiral, 2016-08-25:
@phil-levis that is certainly a valid design choice! Memory allocators in uC are certainly a significant issue. However, having a defragmenting heap allocator could be ultra useful for some applications and completely revolutionize what applications uC can be used in.

I would go both routes -- I would have a "slab allocator" for your drivers and allow your user space to share a dynamic memory region (in addition to the stack, obviously). All user code that relied on heap memory would have to be written in such a way that it would have to wait for it to become available (or you would have to design your system well enough that memory was always available).

You could even have constraints that only allowed certain processes to use so much memory -- that way you might be able to have your cake and eat it too!

alevy, 2016-08-25:
@vitiral that's an interesting suggestion, but not in line with the design of Tock.

  * Processes each get their own dedicated memory regions, which they can do with what they please -- their own stack, own heap, etc.

  * The kernel doesn't have a heap at all. That's an important design decision for us.

  * Drivers can borrow from a process's grant region (which will go away if/when the process dies). That memory is located in a separate region of the process's memory than it's own heap, but ought to be managed like a heap. But they don't their own allocator that is divorce from process memory -- that's very much intentional.

The last one is the only place where I see a use case in Tock for having an efficient (or otherwise) memory allocator in Rust, in the kernel.

alevy, 2016-08-26:
Garret has graciously moved this discussion to a thread on the mailing list here: https://groups.google.com/forum/#!topic/tock-dev/1G9GcqIayL8

vitiral, 2016-08-26:
Gotcha, that is a good design. It sounds like the *processes* might be able to use something like tinymem, since each process get's it's own heap.

Some questions:

# process memory size
Do processes have to define how much memory they will need at compile time?

# grant region
How to you manage when multiple drivers borrow from the grant region? Couldn't you get defragmenetation if:

- A borrowed from grant
- B borrowed front grant
- A stops borrowing (i.e. dies)
- C borrows from the grant

I'm not sure how useful a defragmenting memory manager would be for drivers -- they would have to be willing to suffer pauses when a defragmentation was occuring. In particular, I don't think it would be possible to allow interupts to use tinymem's memory (because the pointer indirection wouldn't work).

Actually, I'm not quite sure why "drivers" are dying at all -- would not driver's be compiled only if they were being used, and if they died that would be really really bad?

# Conclusion
Personally, as a developer, I would rather have a kernel-managed heap than a heap that I have to manage. It is more efficient and I don't have to be QUITE as concerned about how much space I use (although that's a double edged sword, as you point out). It also prevents the need for each process having it's own memory manager, which will be significantly less efficient.

I totally understand why you are making the decision you are though, and wish you the best!

vitiral, 2016-08-26:
I'm copying this onto the email list, please stop discussion here