tock/tock #103: Consider splitting error out of `command_complete` interface

ppannuto, 2016-08-25:
Inspired by https://github.com/helena-project/tock/pull/102/files#diff-6c77447f4906f04d927a4237b1655082L46

Does error as a parameter encourage implementers to simply ignore / not check error? Would we possibly better with separate `command_complete` and `command_error` traits? Sure, people could implement empty error functions, but that seems harder to mess up than forgetting / not bothering to check the error value.

alevy, 2016-08-25:
In general, I would agree that an argument is probably easier to ignore than a whole method you need to implement. The question is basically how cumbersome does it become to code (is there a bunch of duplication, e.g.?).

I think one good way to evaluate this is to just revise all the Client traits that have `_error` arguments to the proposed change and see what that's like.

ppannuto, 2016-08-25:
I *suspect* that it's actually roughly equivalent in code, given that the
code volume of an `if (error != nil) { ... }` is roughly (or exactly)
equivalent to a `def error_handler() { ... }`, but I agree with your
evaluation approach caveated on comparing to implementations that actually
handle errors :D

phil-levis, 2016-08-25:
We went back and forth on this for a long time in TinyOS, with a lot of experiences with both approaches. The eventual conclusion was that having an error parameter was better, often because there is shared code between both successful and failed operations.

phil-levis, 2016-08-25:
That being said, it should be `error`, and NOT `_error`.

alevy, 2016-08-26:
@phil-levis `_error` is just a change that @Idolf made to get rid of warning in @KBaichoo's test code. `_` in Rust, tells the compiler to ignore unused warnings on that parameter. This change is in the client implementation, not the trait.

ppannuto, 2016-08-26:
Right, though I think phil's point (which I'd agree with) is that that
change (the _ only) should be backed out, and there should be a warning
emitted until someone fixes that code / handles that error. Suppressing it
is just asking for trouble :/

I suppose that's the other argument for a separate function is that it
makes it much more obvious when you're just ignoring the error. A single _
is pretty subtle.

@phil-levis, in the shared code case, was it often intermingled, or was it
something that could possibly be addressed by a:

```
callback_success () {
  success_specific_code...
  generic_continue_making_progress();
}

callback_error () {
  success_specific_code...
  generic_continue_making_progress();
}

generic_continue_making_progress () {
  ...
}
```

design pattern?

bradjc, 2016-12-21:
I want to strongly advocate for a single callback. Separate callbacks leads to an annoying amount of code duplication, leads to hard to read code, and (I think) is the less common design pattern overall.

alevy, 2016-12-21:
I think on one of the recent calls we basically agreed with @bradjc about this. Closing this for now, but @ppannuto if you still have comments we can always re-open.

closed 2016-12-21 by alevy