tock/tock #109: Getting started document

ppannuto, 2016-09-19:


ppannuto, 2016-09-19:
How to get hardware, download software, install, etc. How to run tests that your install is correct.
    Details: USB drivers, sload program, etc.
    Necessary: Ubuntu, Mac OS X.
    Plus: Tell people with other Linux installs which packages to install. Linux VM.

bradjc, 2016-12-14:
I think we're close on this one. Still need:
- Link to HW
- VM

alevy, 2017-03-24:
@ppannuto is it still the case that the Vagrant image doesn't help with setting up serial ports, etc?

ppannuto, 2017-03-24:
Correct, I'm not sure what the right thing to necessarily do there is.

The last time I looked into this it looked like Vagrant didn't have a great solution. You can absolutely tunnel them, but you can't configure it from Vagrant itself, have to go into your VirtualBox/VMWare/etc settings. Perhaps the best we can do for now is have a pointer in the Vagrant README to "follow these directions to get USB Serial for your VM environment" :/

alevy, 2017-03-24:
Yeah, I think it's fine to tell people to figure it out based on their VM container. Can we close for now?

commit 6e1cdf6c (2017-03-24): Add a note about USB passthrough

Not perfect, but I don't think there's a better answer for now

Closes #109
closed 2017-03-24 by ppannuto