tock/tock #146: Add Ability to Restart Applications

bradjc, 2016-10-18:
It would be great to be able to restart apps if they crash (or upon demand if the kernel detects they aren't working).

An extra fancy option might be to add the ability to configure the board in one of three modes:

1. `stop`: What we do now, where any hardfault prints the registers and stops.
2. `print_and_reboot`: Print the hardfault error and then restart the MCU.
3. `reboot`: Immediately restart the MCU on any failure.

alevy, 2017-09-28:
This is now implemented as an argument to `load_processes`

bradjc, 2017-09-28:
That is true, but it eventually gets to:

https://github.com/helena-project/tock/blob/4ed764ffd399dcc3d7d344fd7cadf108362ad960/kernel/src/process.rs#L805-L814

which is why this is still open.

alevy, 2017-09-28:
Oh, yep...

bradjc, 2018-05-22:
I'm also thinking that we may want some mechanism to panic the system or provide more information when certain apps crash. For example, we have the `crash_dummy` app to make it possible to print all of the app status/address info on demand, and we may want to continue to allow that, while still auto-restarting other apps.

closed 2018-05-26 by bradjc