tock/tock #178: TakeCell takes many cycles to run

brghena, 2016-11-07:
While debugging why Tock couldn't keep up with UART bytes, I realized that the issue seems to be the amount of time `takecell.map()` is taking when mapping over the App struct in the `nrf51822_serialization` capsule. I believe this is contrary to our expectations that `TakeCell` operations are simple and efficient.

The [takecell-timing branch](https://github.com/helena-project/tock/tree/takecell-timing) has code that demonstrates the issue. To test, program on a firestorm along with the `ble-serialization` app and probe pins P2, P3, and P4. The results are documented inline with comments in the ['nrf51822_serialization` capsule](https://github.com/helena-project/tock/blob/takecell-timing/capsules/src/nrf51822_serialization.rs#L253)

The example that is not commented out uses an instrumented version of `TakeCell` in order to break down where the time is spent. (All instrumentation performed by toggling GPIO pins) The pin toggles can be seen [inline](https://github.com/helena-project/tock/blob/8d407a31523f8ae7fee58e157ae9144c587f8112/capsules/src/instrumented_takecell.rs#L108). The results are: 

| Call | Duration |
|:----|:----------|
| `self.take()` | 10.49 us |
| `maybe_val.map()` | 4.57 us |
| `closure()` | 9.82 us |
| `replace()` | 22.22 us |
| `maybe_val.map()`'s `)};` | 1.33 us |

Attached is the assembly `.lst` file from this example. The code is marked up with `AMIT` on lines where GPIO pins are toggled. **Attachment:** [instrumented_takecell_lst.txt](https://github.com/helena-project/tock/files/574289/instrumented_takecell_lst.txt)

As an example of why this issue is important, when bytes are coming in to the `nrf51822_serialization` capsule at 250000 baud, it has 24 us to process and start a new transaction between bytes. Due to the overhead from `TakeCell`, it takes 48 us to do the processing (TakeCell accounting for 38 us, Closure for 10s). There are workarounds to deal with this (which is how the code is currently implemented), but it would be better to not have to deal with high overhead in the kernel.

**TLDR:** `TakeCell` takes much longer than expected to run and it's complicating drivers by slowing them down.

immesys, 2016-11-07:
FYI, may not be relevant, but on platforms where the number of flash wait states is significant (>1) hotpath code can be optimized by placing it in ram. It is quite useful especially for code used in ISRs because it significantly reduces the jitter of the code execution time (sometimes you hit cache and execute 1 instruction per cycle and sometimes you miss and incur wait states). This recently saved me a lot of trouble in a project, so you may want to consider it for small pieces of library code that get used in ISRs (like TakeCell maybe)

alevy, 2016-11-07:
@branden thanks for doing such a thorough evaluation. There is definitely something fishy going here and I'll take a look when I can tomorrow.

alevy, 2016-11-07:
OK, @brghena's observations are totally correct. The high level problem is that TakeCell _does_ copy the value in/out on a map, except usually what we've been putting in a TakeCell is just a pointer (a mutable reference).

There is a (hypothetical) version of TakeCell that's slightly more general where this wouldn't happen, but at the moment, this is fundamental. Basically, the value inside the TakeCell is actually _stored_ there. At the same time, we allow a reference to an empty take cell to put a new value inside the TakeCell. If the implementation didn't copy value, it would effectively allow someone who has a reference to the empty cell to overwrite an outstanding reference to a value that was previously in the TakeCell.

I've managed to confirm that without doing the take/replace inside of `map`, the overhead is minimal --- the whole operation takes ~8uS while the closure itself takes just under 8uS.

We can get basically similar results by using a flag in the TakeCell for whether something is currently borrowed in a closure (same basic idea as RefCell), but we would have to disallow `take`/`put`/`replace`. This would be slightly less convenient, but the same patters would be workable by putting an `Option` inside the TakeCell.

phil-levis, 2016-11-07:
IIRC, when we first discussed and defined Tock’s execution model, we agreed that interrupts can do non-trivial processing if it’s needed for performance. E.g., the interrupt handler can keep/fill a buffer, and the bottom half uses this buffer. The specific cases that came up when this might be needed are high-frequency ADC sampling and receiving bytes over a bus.

That being said, keeping TakeCell’s overhead low in order to minimize the number of these cases is important.

Phil

brghena, 2017-01-26:
@alevy It's been a little while, and it would be good to revisit this issue.

The main problem I see is unexpectedly using a TakeCell with a value rather than a reference. What I don't understand is why that is occurring in this case and whether it needs to.

If we do not need to ever store values (except for simple data where we are using a Cell), then somehow limiting the existing `TakeCell` to only accept pointers seems like the best choice to me.

Assuming that we need to store a value, my suggestion is that we have two specialized Cell types in Tock: `RefTakeCell` which only accepts pointers to data and `ValueTakeCell` which accepts the value itself. Then it would at least be obvious when a `TakeCell` is going to end up doing a copy.

As a bonus complication, looking through the kernel, there are many places where we use Cell<Option<T>> (usually to hold clients or callbacks) this seems basically equivalent to what I'm thinking of as a `RefTakeCell`, however when it is used on a client it is indeed just storing a reference... ([example](https://github.com/helena-project/tock/blob/c12afb767d389a03e552e5a056e3387a7238ecae/capsules/src/virtual_i2c.rs#L88))

alevy, 2017-01-26:
I'll get to his now, and I basically 100% agree with your suggestion. `RefTakeCell` (our current implementation) is nice because it doesn't impose extra memory cost if the value is `NonZero` (e.g. a pointer), but for exactly the same reason, needs to be moved out of the Option during a `map`. A `ValueTakeCell` would work the same from a user's perspective.

alevy, 2017-02-01:
Preliminary numbers comparing `TakeCell` vs. `MapCell`. For each, I measured a no-op `map` operation in cycle counts as well as size of the data structure with a `&'static u32` and an increasingly large buffer `[[u8; 256]; X]` (for `X` between 1 and 7).

## TakeCell

| Type|Cycles | Bytes|
|-------|------|------|
|`&u32`| 3 | 4|
|`[[u8;256];1]`|1655|257|
|`[[u8;256];2]`|2795|513|
|`[[u8;256];3]`|4365|769|
|`[[u8;256];4]`|4838|1025|
|`[[u8;256];5]`|6780|1281|
|`[[u8;256];6]`|7706|1537|
|`[[u8;256];7]`|10326|1793|

## MapCell

| Type| Cycles | Bytes|
|-------|------|------|
|&u32|18|8|
|[[u8;256];1]|20|257|
|[[u8;256];2]|22|513|
|[[u8;256];3]|20|769|
|[[u8;256];4]|22|1025|
|[[u8;256];5]|22|1281|
|[[u8;256];6]|22|1537|
|[[u8;256];7]|20|1793|

alevy, 2017-02-01:
I think these numbers indicate that `TakeCell` is much better for storing references (half the size as well as fewer instructions), while `MapCell` is better for everything else. In practice, in Tock, this probably means using `TakeCell` for buffers and `MapCell` for other things, like `AppState`, in the i2c example.

brghena, 2017-02-01:
For AppState in [`i2c_master_slave`](https://github.com/helena-project/tock/blob/master/capsules/src/i2c_master_slave_driver.rs#L56) why is AppState not statically defined and then a reference to it stored in a TakeCell?

This is part of my more general confusion of why we are ever storing large objects in `cell`s rather than references to large objects

alevy, 2017-02-01:
@brghena:

First, you'd be wasting a word of storage, since you basically need to references instead of one (one to the TakeCell and another to the `AppState`). Maybe that's not a big deal (only 4 bytes).

Second, it's kind of a bummer from a software engineering perspective. Where do you define/initialize the statically allocated AppState? Currently the only place we really allow it is in the platform configuration, since it's not safe. This means the capsule would have to expose it's internal structure in the interface---the `new` function would have to take an AppState and AppState itself would have to expose of way for an external module to construct it (e.g. make it's fields public).

phil-levis, 2017-02-01:
I'd suggest we call TakeCell RefX and MapCell DataX. Where X is something I haven't figured out. Container is too long. Box already has a meaning. Cell could be OK, but I don't think it explains the use/meaning. Pack is wrong because it's not multiple. Carton sounds like cigarettes. Bin would be mistaken for binary. Case sounds like C. Crate has an existing meaning.

closed 2017-03-24 by alevy