tock/tock #191: HardFault with Tock app sizes larger than 8KB

alevy, 2016-12-02:
[Pasted from an e-mail from @nealjack]

I'm building an app for tock that includes a large library that keeps a lot of static state, and thus a large .bss section. This is on the SAM4L.

I've increased both the app size in my board's main.rs and the SRAM size in userland/linker.ld to 16K. The app compiles to a size of 16106 bytes and flashes successfully, but no code executes, and the processor gets stuck in a low power state.

This issue is complicated by the fact that I do have a working app whose total size is 45671 bytes. So it doesn't seem to be tied to app size directly. This app also has a small .bss size of 40 bytes.

To test if .bss size is the culprit, I've tried adding large static arrays to various other apps, and the size of the .bss section needed to cause the app to fail differs.

The following two build sizes work, but adding one more byte of space to the .bss section causes the weird behavior:

text	  data	   bss	   dec	   hex	
3166   192	    5864	    9222	    2406

text	  data	   bss	   dec	   hex
1432   36	            6108	    7576        1d98

alevy, 2016-12-02:
I was able to replicate this. I got an error that looked like this:

Kernel panic at /home/alevy/hack/helena/tock/chips/sam4l/src/lib.rs:238:
        "Process HardFault.
        r0  0x9db021ba
        r1  0xf66f8485
        r2  0x78ce30a
        r3  0xbc88b216
        r12 0x4dd4ddd0
        lr  0x5174031
        pc  0xfab828cf
        prs 0xf793cd90
        sp  0x20006110
        SHCSR 0x0
        CFSR  0x92
        HSFR  0x40000000

In particular, the last two values (CFSR and HSFR[sic]) indicate a memory management fault.

After some digging, it looks like it's because the size of the MPU region for the process memory is [hard-coded as 8KB](https://github.com/helena-project/tock/blob/391f60619626a4278463058f7a6455d4bdd49589/kernel/src/process.rs#L188). What's worse, it's basically impossible to spot if you aren't the person who wrote the code (me) because the MPU region length is coded as 2^(X+1), so 8KB is 12 --- see line 188 in `kernel/src/process.rs`.

I'm going to fix this right now, but until then you can get around this by just changing `data_len = 12` to `data_len = 13`.

nealjack, 2016-12-17:
Hi Amit,
I've run into more memory issues, maybe related to this. I've found that it matters where large buffers are placed in order to have a working app. I have an app in the Signpost repo that exemplifies this: [https://github.com/lab11/signpost/tree/master/software/apps/mbedtls_ecdh](url). If I move the buffers `buf` and `buf2` into local scope in `main()` the app hard faults. If the buffers are in global scope, the app works fine.

adkinsjd, 2016-12-17:
Notably, this happens even if you increase the stack size by much more than
the size of the buffers.

https://github.com/lab11/signpost/tree/master/software/apps/mbedtls_ecdh

bradjc, 2016-12-18:
I have no idea if this helps, but I've run into what I think are the same issues when testing BLE serialization apps. Trying to do essentially `printf("%s\n, __FILE__);` wasn't working until I disabled the MPU.

adkinsjd, 2016-12-20:
I've done some more exploration into this. If I manually set the app's RAM_LENGTH and change the  process sizes in the kernel, I can successfully access the full range of the process' memory without an MPU fault (as I would expect). This was facilitated by Amit's fix to process.rs. We still however reported problems after the fix:

1) Neal reported a bug above in which moving a buffer into local scope caused a hardfault. There was a bug in the Makefile for the app we were testing on relating to setting the stack size. In my comment above, when I said I was increasing stack size, I was wrong. I fixed the bug, Neal's app now seems to be working with an increased stack size.

2) Sometimes moving buffers from local to global scope causes hardfaults. This was confusing at first, but it actually makes sense (bigger bss + same stack size = hardfault). I think it would still be confusing for a tock user if they caused a hardfault by moving a buffer into global scope, however PR #199 and the addition of flexible app sizes should prevent this scenario.

TLDR; I haven't found any MPU bugs, or bugs generally relating to larger applications. I did find out that moving buffers into global scope can cause a hardfault, but this is not really a bug, and PR #199 should prevent this from happening in the future.

alevy, 2016-12-21:
@adkinsjd thanks for the summary and your and @nealjack's debugging is God's work. I'll close this issue for now, but will also see what might be possible to make this less confusing cause, I agree, this is pretty unintuitive to a user. Any suggestions welcome.

closed 2016-12-21 by alevy