tock/tock #209: nRF51822 Serialization: Issue with receiving packets from nRF

bradjc, 2016-12-18:
If two packets come from the nRF too quickly (as may happen if you are scanning for BLE advertisements in a lab that builds a lot of BLE devices, for instance), the UART receives them as a single buffer, and passes the single buffer to the serialization library. The serialization library likes to do a lot of checking, and throws an assertion if the buffer length does not match what it expects.

In the case of sniffing advertisements, it triggers:

    SER_ASSERT_LENGTH_EQ(index, packet_len);

on line 68 of `ble_gap_evt_adv_report.c`.

alevy, 2017-03-24:
Is this still true @bradjc?

bradjc, 2017-03-25:
I believe this is still true. We kind of get around it now by not using the length that the kernel received, but instead using the length encoded in the header of the packet from the nRF. That way even if there are two concatenated packets, its parsed as if only the first one was received. This works, but isn't great because the second received packet is just dropped.

bradjc, 2017-06-30:
I'm currently trying to fix this. It's turning out to be quite difficult because the serialization library has many assumptions about the hardware that don't necessarily match our model.

closed 2017-07-01 by alevy