tock/tock #226: FXOS8700CQ: chip doesn't seem to go into standby after taking acceleration measurement

bradjc, 2016-12-27:
I'm pretty sure that the FXOS8700CQ driver doesn't actually place the chip into standby after it takes an acceleration measurement.

I think:

```rust
buffer[0] = 0;
self.i2c.write(buffer, 2);
```

is supposed to be:
```rust
buffer[0] = Registers::CtrlReg1 as u8;
buffer[1] = 0; // Set the active bit to 0.
self.i2c.write(buffer, 2);
```

I tried that, but then found that the accel-leds app stopped working. So it's clearly doing something but actually fixing the issue is going to take a little more investigation.

alevy, 2017-03-24:
@bradjc Can you explain how to reproduce this and test it? Is the code You're referencing on line [217](https://github.com/helena-project/tock/blob/5f7246d4af139864f567cebf15bfc0b49e17b787/capsules/src/fxos8700_cq.rs#L217)?

bradjc, 2017-03-25:
Add this diff:
```diff
diff --git a/capsules/src/fxos8700_cq.rs b/capsules/src/fxos8700_cq.rs
index 5f7ab422..8682704c 100644
--- a/capsules/src/fxos8700_cq.rs
+++ b/capsules/src/fxos8700_cq.rs
@@ -214,7 +214,8 @@ impl<'a> I2CClient for Fxos8700cq<'a> {
                 let y = ((y as isize) * 244) / 1000;
                 let z = ((z as isize) * 244) / 1000;
 
-                buffer[0] = 0;
+                buffer[0] = Registers::CtrlReg1 as u8;
+                buffer[1] = 0; // Set the active bit to 0.
                 self.i2c.write(buffer, 2);
                 self.state.set(State::ReadAccelDeactivating(x as i16, y as i16, z as i16));
             }
```

Then flash that kernel. I use the `hail` app and what I see is that everything else keeps working, except the acceleration value never changes.

The code as it is in the repo today is supposed to be "deactivating" the accelerometer (give the state names), but is writing some value to the status register, which I don't think does anything.

brghena, 2017-05-20:
Closed by #358

closed 2017-05-20 by brghena