tock/tock #229: Hail: mac port arm-none-eabi-gcc 5.1.0 fails to compile program

cxbrooks, 2016-12-29:
The [Getting Started](https://github.com/helena-project/tock/blob/master/doc/Getting_Started.md) guide already states that Brew is preferred over Mac Port, but it should probably be edited to state that Mac Port is not supported, or at least arm-non-eabi-gcc 5.1.0 is not supported when trying to program the Hail.

In tock/boards/hail, with /opt/local/bin in my path and `sudo port upgrade outdated` recently run, `make program` fails with  `ld: cannot find crt0.o: No such file or directory`      

Below is a sample run.

```
bash-3.2$ which arm-none-eabi-gcc                                                                                                 
/opt/local/bin/arm-none-eabi-gcc                                                                                                  
bash-3.2$ /opt/local/bin/arm-none-eabi-gcc --version  
arm-none-eabi-gcc (GCC) 5.1.0                                                                                                     
Copyright (C) 2015 Free Software Foundation, Inc.                                                                                 
This is free software; see the source for copying conditions.  There is NO  
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
                                                                                                                                  
bash-3.2$ make clean                                                                                                              
bash-3.2$ make PORT=/dev/tty.usbserial-00002014 program     
   Compiling hail v0.1.0 (file:///Users/cxh/src/hail/tock/boards/hail)  
   Compiling rust-libcore v0.0.3                                                                                                  
   Compiling kernel v0.1.0 (file:///Users/cxh/src/hail/tock/kernel)      
   Compiling cortexm4 v0.1.0 (file:///Users/cxh/src/hail/tock/arch/cortex-m4)   
   Compiling capsules v0.1.0 (file:///Users/cxh/src/hail/tock/capsules)       
   Compiling sam4l v0.1.0 (file:///Users/cxh/src/hail/tock/chips/sam4l)   
error: linking with `arm-none-eabi-gcc` failed: exit code: 1                                                                      
note: "arm-none-eabi-gcc" "-mcpu=cortex-m4" "-mthumb" "-mfloat-abi=soft" "-Wl,-gc-sections" "-Tlayout.ld" "-L" "/Users/cxh/.multirust/toolchains/nightly-2016-07-29-x86_64-apple-darwin/lib/rustlib/sam4l.json/lib" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/hail.0.o" "-o" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/hail" "-Wl,--gc-sections" "-Wl,-O1" "-nodefaultlibs" "-L" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps" "-L" "/Users/cxh/.multirust/toolchains/nightly-2016-07-29-x86_64-apple-darwin/lib/rustlib/sam4l.json/lib" "-Wl,-Bstatic" "-Wl,-Bdynamic" "-lm" "-lgcc" "-lc"                      
note: /opt/local/lib/gcc/arm-none-eabi/5.1.0/../../../../arm-none-eabi/bin/ld: cannot find crt0.o: No such file or directory      
collect2: error: ld returned 1 exit status                                                                                        
                                                                                                                                  
error: aborting due to previous error                                                                                             
error: Could not compile `hail`.                                                                                                  
                                                                                                                                  
To learn more, run the command again with --verbose.                                                                              
make: *** [target/sam4l/release/hail.elf] Error 101                                                                               
bash-3.2$ make -n                                                                                                                 
cargo build --release --target=sam4l.json                                                                                         
cp target/sam4l/release/hail target/sam4l/release/hail.elf                                                                        
arm-none-eabi-size target/sam4l/release/hail.elf                                                                                  
arm-none-eabi-objcopy -Obinary target/sam4l/release/hail.elf target/sam4l/release/hail.bin                                        
bash-3.2$ cargo build --release --target=sam4l.json                                                                               
cargo build --release --target=sam4l.json                                                                                         
   Compiling hail v0.1.0 (file:///Users/cxh/src/hail/tock/boards/hail)                                                            
error: linking with `arm-none-eabi-gcc` failed: exit code: 1                                                                      
note: "arm-none-eabi-gcc" "-mcpu=cortex-m4" "-mthumb" "-mfloat-abi=soft" "-Wl,-gc-sections" "-Tlayout.ld" "-L" "/Users/cxh/.multirust/toolchains/nightly-2016-07-29-x86_64-apple-darwin/lib/rustlib/sam4l.json/lib" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/hail.0.o" "-o" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/hail" "-Wl,--gc-sections" "-Wl,-O1" "-nodefaultlibs" "-L" "/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps" "-L" "/Users/cxh/.multirust/toolchains/nightly-2016-07-29-x86_64-apple-darwin/lib/rustlib/sam4l.json/lib" "-Wl,-Bstatic" "-Wl,-Bdynamic" "-lm" "-lgcc" "-lc"                      
note: /opt/local/lib/gcc/arm-none-eabi/5.1.0/../../../../arm-none-eabi/bin/ld: cannot find crt0.o: No such file or directory      
collect2: error: ld returned 1 exit status                                                                                        
                                                                                                                                  
error: aborting due to previous error                                                                                             
error: Could not compile `hail`.                                                                                                  
                                                                                                                                  
To learn more, run the command again with --verbose.                                                                              
bash-3.2$  cargo build --verbose --release --target=sam4l.json                                                                     
cargo build --verbose --release --target=sam4l.json                                                                               
       Fresh rust-libcore v0.0.3                                                                                                  
       Fresh kernel v0.1.0 (file:///Users/cxh/src/hail/tock/kernel)                                                               
       Fresh cortexm4 v0.1.0 (file:///Users/cxh/src/hail/tock/arch/cortex-m4)                                                     
       Fresh capsules v0.1.0 (file:///Users/cxh/src/hail/tock/capsules)                                                           
       Fresh sam4l v0.1.0 (file:///Users/cxh/src/hail/tock/chips/sam4l)                                                           
   Compiling hail v0.1.0 (file:///Users/cxh/src/hail/tock/boards/hail)                                                            
     Running `rustc src/main.rs --crate-name hail --crate-type bin -C opt-level=3 -C panic=abort -C lto --out-dir /Users/cxh/src/hail/tock/boards/hail/target/sam4l/release --emit=dep-info,link --target sam4l.json -L dependency=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps --extern sam4l=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libsam4l.rlib --extern capsules=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcapsules.rlib --extern core=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcore-882297ed0c39d543.rlib --extern kernel=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libkernel.rlib --extern cortexm4=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcortexm4.rlib`       [... cut ...] elease/deps/libsam4l.rlib --extern capsules=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcapsules.rlib --extern core=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcore-882297ed0c39d543.rlib --extern kernel=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libkernel.rlib --extern cortexm4=/Users/cxh/src/hail/tock/boards/hail/target/sam4l/release/deps/libcortexm4.rlib` (exit code: 101)   
bash-3.2$
```

I was able to compile program using brew.

ppannuto, 2017-01-04:
> I unfortunately can't find a arm-none-eabi-gcc 5.1 binary for Linux to try
> to compare, but I may be able to find a Mac with macports to investigate
> further. In the mean time I have emailed the macports Portfile maintainer
> to ask if he can update the version to 5.4 which I would expect to take
> care of this going forward.

We may just want to drop the macport recommendation altogether.

Years ago port had support for more packages and there was a reason to
mention it. Nowadays it's just worse than brew. I don't think it's worth
our time or effort to try to support macports.

cxbrooks, 2017-01-04:
Unfortunately, the GDP does not work well with brew, so the reality is that both are needed.

Updating the docs to tell people to use brew seems reasonable.  Instead of removing macports entirely from the docs, you might want to state that macports with arm-none-eabi-gcc 5.1 does not work.

brghena, 2017-01-04:
Agreed. I'm updating the Getting Started readme right now to include this.

brghena, 2017-01-04:
Brad's PR to macports just went through. https://github.com/macports/macports-ports/blob/master/cross/arm-none-eabi-gcc/Portfile

@cxbrooks would you be willing to test this and make sure that it works for Tock?

cxbrooks, 2017-01-05:
Sure, I can test the Macports update of arm-none-eabi-gcc when it is published. 

I'm not sure how Macports does updates.  I did 

```
sudo port selfupdate
sudo port upgrade outdated
```

However, /opt/local/bin/arm-none-eabi-gcc is still 5.1.

Presumably, there is a release process here somewhere that needs to happen.

phil-levis, 2017-01-05:
"Years ago port had support for more packages and there was a reason to mention it. Nowadays it's just worse than brew. I don't think it's worth our time or effort to try to support macports."

Generally speaking, a good rule of thumb is that if someone is willing to maintain it, then it's worth our time. However, this means that if someone is willing to maintain it, then other people need to test it when they make changes (we can't ask that person to clean up other people's messes). 

I agree that we're not yet at the point where we want to maintain installation approaches we don't ourselves use (TinyOS cygwin was this way for a long time, no developer used it but we still maintained it).

closed 2017-02-13 by ppannuto

skade, 2017-04-16:
I'm not much of an embedded software developer, but isn't the problem here that macports explicitely disables building newlib and this ships without crt0?

https://github.com/macports/macports-ports/blob/master/cross/arm-none-eabi-gcc/Portfile#L19

commit bffa0b9e (2017-02-13): userland: validate that the compiler is new enough

Closes #229.