tock/tock #235: Fix Apps and MPU Dynamic Sizing in Memory

bradjc, 2017-01-05:
The ARM MPU requires that regions start on multiples of the region size. Right now, we just make each app use a power of two in memory and everything works out (this is also how it worked before #199). But, to make apps actually dynamically sized and not just have a massive amount of heap space, the grant section at the end needs to be positioned so that its minimum length is met but that it also starts on a multiple of its size. This is then going to cause the next application to maybe not start on a power of two, so the kernel must take care to position its start at a multiple of its MPU region size.

Note: failing to do this seems to cause really strange faults in the MPU (that occur immediately when the app tries to run).

brghena, 2017-01-06:
One notable aspect of the way things are currently working in master is that extra unallocated space (created by making the app the next power of two big) is left between the grant and heap regions. This allows them to overrun their bounds a little without necessarily faulting. In the `tests/hail` and `sensors` app, for example, the heap was growing to 1600 bytes while only being allocated 1024.

One idea that could be included into Tock is to check these bounds at each syscall and immediately fault apps that violate them. This might help avoid issues where future unrelated changes to the app make it actually run out of heap space.

alevy, 2017-09-28:
We ended up fixing this a while ago. It's maybe not optimal still, but if nothing else this description is stale. Closing for now.

closed 2017-09-28 by alevy