tock/tock #242: userland build needs to handle ~ expansion

ppannuto, 2017-01-12:
@srohrer32 had

```
 3 TOCK_DIR = ~/git/tock                                                           
 4 TOCK_USERLAND_BASE_DIR = $(TOCK_DIR)/userland 
```

in the Makefile for his out-of-tree app, which caused all kinds of sadness like `/home/srohrer/git/embeddedMFCC/app/~/git/tock/userland/libtock/Makefile:`

It looks like neither `abspath` nor `realpath` in make expands tildes, so need to figure something out here

alevy, 2017-01-12:
Does `~` ever work in Makefiles?

ppannuto, 2017-01-12:
No idea, just documenting the issue for me to look at later.

My 5m of searching indicated that make doesn't play nicely with ~, but at
the same time, what Sam did doesn't seem unreasonable, it'd be good to find
a way to either support it, or at least detect / warn / fail if there's a ~

alevy, 2017-01-13:
Got it, makes sense.

It seems reasonable to try and do the best possible within Makefile's capabilities, but if Makefile don't generally support it, my vote is against outsmarting it.

I don't know for sure either, I just know I've gotten in the habit of specifying `$(HOME)/...` when I pass variables to make.

PR #250 (2017-01-17): userland: Check for ~ in TOCK_USERLAND_BASE_DIR
Errors if someone tries to use `~` instead of `$(HOME)`.

Fixes #242 

Seemingly confirmation that using ~ is just bad: https://github.com/sudar/Arduino-Makefile/issues/320
commit 2b0f8e64 (2017-01-17): userland: Check for ~ in TOCK_USERLAND_BASE_DIR

Errors if someone tries to use ~ instead of $(HOME).

Fixes #242
closed 2017-01-17 by alevy