tock/tock #252: Disable or warn on use of panic'ing `[]` in kernel?

ppannuto, 2017-01-17:
As noted by @alevy on #251, the `[]` operator can panic, which we'd generally prefer the kernel not do. Given that you can (almost?) always work around using it, perhaps we should find a way of preventing kernel code from using the `[]` operator? Maybe this is something we can `#feature` around, or maybe as a plug-in to rustfmt?

I'm especially concerned about this for capsule authors, since it's probably not obvious to the average programmer that the array index / slice operator can panic.

bradjc, 2017-01-17:
I agree, if this is possible. I don't know how to tell the difference between https://github.com/helena-project/tock/blob/36486c511b9b429eb1844e01e7e352a5fe8e5b35/capsules/src/rng.rs#L66 and the bad one, however.

Also there are a bunch of spots where this would have to be fixed in the existing capsules.

alevy, 2017-01-17:
I don't know that it's possible to disable this without modifying Rust core... `[]` is just sugar for the `Index` and `IndexMut` trait's `index` and `index_mut` methods (respectively). It's the implementation of those for the slice primitive which panics.

Also, turns out I was wrong, and in fact it does panic in some cases (this is an implementation decision that I don't necessarily agree with cause it can always safely return an empty list). I only tested an edge case that happens to work (`myslice[myslice.len()..]` returns an empty list).

alevy, 2017-01-17:
instead of doing an index, in cases where it would be reasonable, it's possible to instead use the `get(&self, usize)` method of slice, which returns an `Option<T>` and doesn't panic.

phil-levis, 2017-01-23:
This seems like a good “style guide” recommendation to have, but should not be something we enforce. Every hoop you add makes it harder to for someone to contribute.

Phil

ppannuto, 2018-05-08:
I think this is something we'd love to have, but would require upstream support to do in a sane way, and not a top priority.

Closing for now. If this becomes easier upstream in the future, happy to re-open and re-consider.

closed 2018-05-08 by ppannuto