tock/tock #258: Clean up lingering `ReturnCode::FAIL`s

ppannuto, 2017-01-22:
I feel like `ReturnCode::FAIL` is kind of an unhelpful cop-out, and the kernel should avoid using it if possible in favor of a more specific error. After #256, there are a few places where there are currently `FAIL`s that should be replaced:

 - User called subscribe without first calling allow
    - [console](https://github.com/helena-project/tock/blob/f725aaa6e2c20ad7f66fdbf24984a20ba885b6e9/capsules/src/console.rs#L146)
    - [nrf serialization](https://github.com/helena-project/tock/blob/f725aaa6e2c20ad7f66fdbf24984a20ba885b6e9/capsules/src/nrf51822_serialization.rs#L173)
 - Two others that look similar
    - [nrf serialization](https://github.com/helena-project/tock/blob/f725aaa6e2c20ad7f66fdbf24984a20ba885b6e9/capsules/src/nrf51822_serialization.rs#L160)
    - [spi](https://github.com/helena-project/tock/blob/f725aaa6e2c20ad7f66fdbf24984a20ba885b6e9/capsules/src/spi.rs#L199)

phil-levis, 2017-01-23:
FAIL is only a unhelpful cop-out if its meaning isn't documented. I.e., allowing some uses of FAIL is, IMO, more useful than a huge proliferation of weird edge case values.

ppannuto, 2017-01-23:
That's true, but it does feel like reporting something (maybe `EACCESS`) for the subscribe but forgot allow case is common enough that it's worth handling. Less sure about the second two.

alevy, 2017-01-23:
Where would it be reported? I'm not sure it's worth enforcing a particular ordering of `subscribe` and `allow`, and without an ordering, it's not clear to me where the error would show up. Isn't it the case that `subscribe` can always work even if a buffer hasn't been shared? It might be kind of useless, but leave it up to the application to decide if it's useful.

ppannuto, 2017-01-23:
Isn't there already an ordering of allow and subscribe?
https://github.com/helena-project/tock/blob/f725aaa6e2c20ad7f66fdbf24984a20ba885b6e9/capsules/src/console.rs#L146

Unless I misunderstand the code, currently, if you call subscribe without ever having called allow, then you will `enter` successfully, but the `app.write_buffer` will be empty, and we'll currently return `FAIL`.

Is there another scenario where that will return `FAIL`? Ohh... just realized, if you call subscribe twice in a row without waiting for the callback, then the usart driver may still own the `app.write_buffer`, so it will be empty in that case too. Hmmm.. interesting. Those are two very different errors that are harder to distinguish. Originally I'd thought it was only the first scenario that could get to that line.

closed 2017-05-31 by bradjc