tock/tock #268: i2c master slave driver breaks tock

nealjack, 2017-01-31:
Adding the i2c_master_slave_driver to hail kernel makes the device unresponsive - no serial output and no panic LED flashing.

I made a branch that exemplifies this: 53e5755

bradjc, 2017-01-31:
change boards/kernel_layout.ld:

    __stack_size__ = DEFINED(__stack_size__) ? __stack_size__ : 0x2000;

Seems to fix it.

nealjack, 2017-01-31:
Is this something we should change in tock master?

alevy, 2017-01-31:
Does it also fix things if `platform` is statically allocated instead of on the stack? If so, how big is `platform` (last argument you'd need to give `static_init!`?

alevy, 2017-01-31:
@nealjack I think there is nothing wrong with increasing the stack size, so sure, increasing in master seems fine. but wanna get a handle on why the stack is so large...

nealjack, 2017-01-31:
@alevy the i2c_master_slave driver currently requires 116 bytes. Is that what you wanted to know?

alevy, 2017-01-31:
It was, but then I realized that's not the important question and I also realized I could answer it on my own. `hail` is allocated on the stack, but it's only pointers and is 56 bytes... I'm basically trying to figure out where 4096 bytes are going.

alevy, 2017-01-31:
it looks like `i2c_master_slave` is consuming something like 1.7KB of stack. It's not super obvious to see this, but comment out it's case in [`with_driver` on `boards/hail/src/main.rs`](https://github.com/helena-project/tock/blob/53e575506e29d9d8e3428caba1249c97ff30a209/boards/hail/src/main.rs#L107) and see how the stack decreases in second instruction of `reset_handler` in the output assembly (from 4000 bytes to around 2300 bytes).

You might ask, why is `i2c_master_slave` using so much stack?

TakeCell isn't helping. AppState (which is stored in a `TakeCell`) is 76 bytes, and every time we call `TakeCell#map` (I believe) it's making room on the stack to copy that whole AppState out. This is exacerbated by the fact that the fields of `AppState` are also, needlessly `TakeCell`s (you only ever need one level of `TakeCell`).

This doesn't totally explain why we're getting 1.7KB of stack consumption from the `Driver` methods, since none of them call subroutines and each only every uses `AppState` once. My best guess is that the compiler is not smart enough to eliminate the stack allocation for each branch of match statements, even though those branches are totally independent. This is partially confirmed by moving `app_state.map` to the outside of the map in `allow`.

Replacing the fields inside the AppState with Option instead of TakeCell (you only ever need one level of TakeCell) saves _something_ but not a ton (about a 100 bytes total).

Replacing `TakeCell<AppState>` with a [`MapCell`](https://github.com/helena-project/tock/blob/perf/takecell/kernel/src/common/take_cell.rs#L139) (work in progress still) eliminates the problem since `MapCell` doesn't copy out on a call to `map`.

For example, the biggest culprit,`Driver#command`, is down to 24 bytes from ~1KB.

I bet we can get significant stack space back in other cases too with MapCell.

# High bit

Increasing the stack size is a fine fix for now to make progress, but `MapCell` will fix this better, and a stack size larger than 4KB is alarming. @brghena's suggestion of trying to protect from stack overflows in the kernel with the MPU makes sense too.

alevy, 2017-01-31:
I pushed my code to git://github.com/alevy/tock on the `i2c-ms-hail` branch, but it should probably only be used for testing as I still need to finish `MapCell`.

bradjc, 2017-02-01:
Could `TakeCell`'s implementation get replaced with `MapCell`'s implementation?

alevy, 2017-02-01:
Preliminary numbers comparing `TakeCell` vs. `MapCell`. For each, I measured a no-op `map` operation in cycle counts as well as size of the data structure with a `&'static u32` and an increasingly large buffer `[[u8; 256]; X]` (for `X` between 1 and 7).

## TakeCell

| Type|Cycles | Bytes|
|-------|------|------|
|`&u32`| 3 | 4|
|`[[u8;256];1]`|1655|257|
|`[[u8;256];2]`|2795|513|
|`[[u8;256];3]`|4365|769|
|`[[u8;256];4]`|4838|1025|
|`[[u8;256];5]`|6780|1281|
|`[[u8;256];6]`|7706|1537|
|`[[u8;256];7]`|10326|1793|

## MapCell

| Type| Cycles | Bytes|
|-------|------|------|
|&u32|18|8|
|[[u8;256];1]|20|257|
|[[u8;256];2]|22|513|
|[[u8;256];3]|20|769|
|[[u8;256];4]|22|1025|
|[[u8;256];5]|22|1281|
|[[u8;256];6]|22|1537|
|[[u8;256];7]|20|1793|

alevy, 2017-02-01:
I think these numbers indicate that `TakeCell` is much better for storing references (half the size as well as fewer instructions), while `MapCell` is better for everything else. In practice, in Tock, this probably means using `TakeCell` for buffers and `MapCell` for other things, like `AppState`, in the i2c example.

ppannuto, 2017-02-01:
Alternatively, when/why do we ever need to pass around the whole object in
the cell? Why don't we just use references for everything everywhere? [or
is the effective result that mapcell takes an object and stores a reference
instead?]

Can the trait signatures enforce best practices (i.e. can TakeCell<&T> and
MapCell<T> enforce only refs in the former and only non-refs in the latter)?

alevy, 2017-02-01:
@ppannuto yes, to enforcing best practices.

Re: first question, see my response to Branden in #178 (which I think is where we should continue the discussion on MapCell/TakeCell to consolidate)

alevy, 2017-02-01:
On advice from @rkruppe on the rust internals IRC, I tried recompiling with a recent nightly version of Rust (apparently the libtrans package of the compiler that translates from MIR to LLVM IR has been improved substantially since July). It seems to basically fix the problem---stack usage is down in the new version to ~1.2K total as opposed to over 4KB.

This doesn't negate the switch to `MapCell` vs `TakeCell` proposed in #178 since, but it does explain why the impact was so large, I think.

brghena, 2017-02-01:
Oooh, can we upgrade Rust versions again?! One of our undergrads accidentally had the newest nightly of Rust for a bit. Tock didn't build right, but when it printed errors they were formatted much more nicely than the current nightly. Very readable

alevy, 2017-02-01:
Yeah, the new Rust nightly has great error messages! I'm for it, it's a very very easy upgrade. We should do it probably after merging #261 ?

ppannuto, 2017-02-21:
I think this issue has been resolved?

closed 2017-03-24 by alevy