tock/tock #270: Should we add a `bug_on!` or some such?

ppannuto, 2017-02-01:
We have a bunch of stuff like this https://github.com/helena-project/tock/pull/261/files#diff-fff017e1556abaa2677b58fe9aa9935bR552 where we never expect to get to that state, but don't want to burn the world down in normal operation.

I think some macro that passes by silently in a release build, but panic's in a debug build could be a useful thing to add. Anything obviously flawed there?

ppannuto, 2017-02-01:
Related:

[image]

brghena, 2017-03-01:
As discussed on Monday, a happy middle ground here might be [expect()](https://doc.rust-lang.org/std/option/enum.Option.html#method.expect)

However, that still panics all the time, not just in debug compiles.

ppannuto, 2017-03-01:
I actually think both `bug_on` and `expect` could be useful, though clearly elucidating when to use each might be a little harder.

bradjc, 2017-03-01:
If there's a clear case for `bug_on`, great, otherwise I'm always for sticking with things that rust developers would already be familiar with.

closed 2017-03-24 by alevy