tock/tock #274: SPI over USART CS line deasserted too early

brghena, 2017-02-02:
In the current implementation of SPI over USART, the chip select line is deasserted too early. The `chips` driver calls `cs.set()` (as chip select is a GPIO line) when it receives the `xfer_done` callback from the DMA. Unfortunately, even once the DMA is done there may still be bytes on the wire in the case of a large (greater than ~10 bytes) transfer.

As seen in the attached image, the CS line is deasserted about 1.5 bytes too early in an 18 byte transfer. The problem occurs [here](https://github.com/helena-project/tock/blob/c9aa9a14ad332b2f4be451ea6f6969562eea8a19/chips/sam4l/src/usart.rs#L493). As noted a few lines lower in a comment, the transfer has not actually finished on the wire yet.

[image]

A real fix for this is going to have to identify an appropriate interrupt signal from the USART that can act as a trigger for deasserting the chip select. For the moment, an easy band-aid is to increase the transaction length of `read_write_bytes(...)` calls by 2 for drivers that expect to run on top of the USART. For most hardware peripherals, reading two extra bytes is not harmful, so a thus-modified driver should still work on top of a true SPI.

phil-levis, 2017-02-02:
You need the read interrupt, not the write interrupt. 

Phil

alevy, 2017-03-24:
From #277:

> The second bullet goes part-way to fixing issue #274. In the case of a
SPI transaction that does a read, the CS line will now be deasserted
properly. In the case of a many-byte write with no read, the problem is
not solved (but has not been made worse either).

commit 111bf988 (2018-06-16): sam4l: usart: spi: fix callback location

The SPI driver had the same issue as the UART driver: the callback was
triggered not when TX was in its Idle state, but instead when it was in
its Transfer Completing state. This means if you do anything in the
callback routine things will break.

This should also address the issue with the CS line going back high too
early, since now it is not set until it is safe to issue the callback,
which should definitely be at the end of the transaction.

Fixes #274
closed 2018-06-16 by alevy