tock/tock #290: Improvement: More robust timer in libtock

bradjc, 2017-02-18:
Eventually we need to upgrade `timer.h` to allow multiple timer intervals and callbacks on top of the virtual timer in the kernel.

alevy, 2017-02-19:
This might imply that we want the driver to be an alarm, not a timer, because timers are difficult to visualize.

adkinsjd, 2017-06-03:
Has there been any further thought to this issue? I would really like to add timeouts (or even delays) to some of our userland libraries, which isn't possible in the current timer implementation.

alevy, 2017-06-04:
Yeah, doing this in userland at least should be pretty straight forward. I'll give it a shot tonight.

alevy, 2017-06-04:
@adkinsjd can you remind me if you're using `timer_read` anywhere? I can't find any references in the Signpost repo. If you do, can you remind me if it's important for the resolution returned to be finer grained than milliseconds?

brghena, 2017-06-04:
The microwave radar and a synchronization test use it. https://github.com/lab11/signpost-software/search?utf8=%E2%9C%93&q=timer_read&type=

Also, FYI, we've moved all software to the `signpost-software` repo so that you can clone it at a remotely reasonable size.

brghena, 2017-06-04:
Microwave Radar just wants a fine-grained as it can get. It could make due with milliseconds if it had to. (future versions would probably switch to continuous ADC sampling at a set frequency anyways)

adkinsjd, 2017-06-04:
We would like the timer_read function to be as accurate as possible so that we can keep timing between PPS signals. That being said, keeping good time between PPS doesn't really work in Tock for multiple reasons, so I don't think it would be too problematic if timer_read only returns time at millisecond granularity.

alevy, 2017-06-04:
@adkinsjd oh, for calibrating against GPS? yeah, you ought to do that in the kernel anyway to minimize jitter (and probably want to use one of the TIMER hardware controllers on the SAM4L instead of a virtualized AST or something).

@brghena and @adkinsjd OK, so i have leeway to make the timer_read function return only milliseconds (instead of whatever the kernel driver sees, which is 16kHz on the SAM4L), but if I can get away with virtualizing the timer without doing that (say by introducing a call to get the frequency), that would be preferable, yeah? (thumbs up for yes, thumbs down for no)

adkinsjd, 2017-06-04:
Yeah it should be in the kernel (assuming the SAM4L can even support timing we need at all - I'm skeptical)

I think it's fine if the timer_read function returns only milliseconds.

alevy, 2017-06-04:
started work on this here: <https://github.com/helena-project/tock/compare/master...alevy:feature/user/virtalarm>

High level, I added commands to the Timer system-call driver. One that returns the frequency of the clock and another that sets an alarm for an absolute tic value. Then, I started building a userland virtual alarm on top of those calls. It turns out it be much trickier (slash maybe infeasible) to virtualize a straight up timer rather than an alarm (we already kind of knew this).

The virtual alarm in userland in nearly complete except it doesn't account for clock wrapping yet (just some annoying math to get right and also only happens once every 8 years on the SAM4L), and it's also totally untested.

PR #419 (2017-06-22): Userland virtual alarm
Fixes #290 

This PR adds support for multiple concurrent alarms (not timers, but we can implement timers on top of this) in userland. It does so with a small extention to the Timer system call driver and a userland library that uses a binary heap to keep track of outstanding alarms.

In order, the commits do the followin:

  1. Add commands to the Timer system-call driver. One that returns the frequency of the clock and another that sets an alarm for an absolute tic value.

  2. An initial implementation of a userland virtual alarm on top of those calls. It turns out it [... cut]
closed 2017-06-22 by alevy