tock/tock #311: How to use create_libnrfserialization.sh script

jmbrederveld, 2017-03-08:
If I call the script directly I get the following errors:

Makefile:6: *** TOCK_ARCH not set.  Stop.
cp: cannot stat 'build/cortex-m4/libnrfserialization.a': No such file or directory
Makefile:6: *** TOCK_ARCH not set.  Stop.
cat: 'build/cortex-m4/*.headers': No such file or directory

Do I need to manually set these environment variables?

alevy, 2017-03-08:
@bradjc I think this one's for you, but it seems so.

we include a pre-compiled libnrfserialization.a in the repo, so normally you don't have to do this.

If I set the environment variables to:

```bash
export TOCK_ARCH=cortex-m4 # This could also be cortex-m0 or cortex-m3, it's just passed as is to GCC
export BUILDDIR=build
```

It successfully builds a version of libnrfserialization.a in the `build` subdirectory.

However, in general, this library is pretty specific to Tock (it basically just implements the NRF serialization library stubs with system calls to the Tock kernel) and not necessarily something we'll keep around long term when we have more native support for BLE on the nrf51 (it pretty fundamentally means the NRF can't go into low power). So just beware, your mileage may vary. If you're planning on using this outside of a Tock platform, it's probably a decent reference implementation, but you're probably better off writing your own based on the NRF SDK serialization library directly.

ppannuto, 2017-03-08:
Quick skim on phone of thread: probably my fault actually. Makefile used to
set TOCK_ARCH to a default of cortex_m4, now it doesn't, picks based on
board, and the other pre built libs have a built image for each arch. Will
fix.

closed 2017-03-24 by alevy