tock/tock #315: Questions: STM32 support and other

stephanbuys, 2017-03-16:
Hi all, I'm very new to the embedded space and have just recently 'inherited' an important project that involves creating an embedded device, probably based on Cortex-M. Being completely new to the space we went ahead and ordered some dev boards which turned out to be STM32F4x based.

@japaric has done a lot of work on chips based on that platform over at https://github.com/japaric

Being a total n00b for all practical purposes I was wondering if someone here might chime in to the following questions:
1) Should we just pick a board that is already supported by tock (if it is feasible to use tock, see below), and base our device on that? (we are looking at 30,000+ devices).
2) If not, what does it take to "port"/implement a HAL to support devices like the STM32F3/F4? (or other arbitrary Cortex-M based microprocessors?)
3) Can tock be used to:
   - Communicate as a WiFi client (probably using something like a ESP8266 serial wifi module)
   - Be a TCP client
   - run some light crypto (SHA256)
   - retain some state (unique identifier, counters, etc)

Tock looks super exciting!

Thanks for your time.

phil-levis, 2017-03-16:
Stephan,

I’m glad you’re very interested in Tock. It’s currently supported on the SAM4L series of Atmel CortexM-4s, and there’s growing support for the nRF51 series of CortexM-0 SoCs that have an integrated Bluetooth Low Energy transceiver. We chose these two parts because they technically seem to have the best energy/power tradeoffs, and so from a research standpoint allow us to push the frontier of what’s possible.

That being said, the core development team agrees that the STM32 series should be our next supported micro controller family. No one has yet volunteered to start on this — we’re currently focusing on providing some higher level services on the existing kernel (e.g., 6lowpan, power management). STM has a bunch of nice properties, particularly its price point. It would be great to run Tock on Nucleo boards.

So, if you’d like to port Tock to an STM32, I think you’ll find people very willing to help and answer questions when they come up, to support you in the effort. The scale of the effort depends on how much of the HAL you want to implement. I think the general minimal system is booting (main.rs), GPIO, UART, and timers. Of these, usually booting and timers are the trickiest, the former because if it’s not working debugging is tricky and the latter because of a few edge cases. We’ve found that then slowly growing the HAL as peripherals are needed (SPI master/slave I2C master/slave, ADC, CRC, AES, etc.) works well.

As for boards currently supported by Tock, they are 4 research boards (firestorm, imix, hail, signpost) and the nRF51DK. We are working on making imix publicly available, but it will be at least two months before that happens. imix has a SAM4L application MCU, an nRF51422 for BLE, an Atmel RF233 for 802.15.4, and a custom circuit we designed for highly dependable random number generation. It has lots of pinouts, e.g., for measuring power. 

So my suggestion for whether to go with a SAM4L-based design or an STM32-based design boils down to money vs. time. If you want your devices to be as cheap as possible, then bringing up an STM32 in Tock is probably the best bet. If you’re willing to use a more expensive MCU for the purpose of saving developer time, then I’d suggest going with a SAM4L. If you go the SAM4L route, then it should not be difficult to bring up a development kit, and we can help you with the effort.

As for what Tock can do,

  - Tock can communicate over the serial port to other chips, so you could build a board that connects the Tock MCU to an ESP8266. For example, this is how the SAM4L interacts with the nRF51 on Firestorm and imix. We do not have an ESP8266 driver, though, you would have to write that. 
  - Tock currently does not have TCP support. My hope is that it will by the end of the summer.
  - Tock can run crypto, with two caveats. First, there currently isn’t an implementation for SHA256, but you could take one in C or assembly and call it from Rust, or port SHA256 to Rust. Second, the one thing the kernel can’t do is run long-running computations, so you might need to put it in a process. A lot of MCUs today include crypto accelerators, if you find one that has the ciphers you want, that’s the best choice (both for power and for time).
  - Tock can be used to retain some state. Do you mean in non-volatile storage? We have some initial interfaces for the kernel to read/write flash storage on the SAM4L. If you mean in RAM, sure, most definitely, no problem.

Phil

kfihihc, 2017-03-19:
Tock for STM32 +1,

But it seem that is not any guide to help people to port it?

BTW, I recommand this board: [BLE Carbon](http://wiki.seeed.cc/BLE_Carbon/) (STM32F401RE + nRF51822 32K RAM, 256K Flash). But the STM32F401 connect nRF51822 with SPI.

[image]

alevy, 2017-03-19:
@kfihihc Indeed, that'll probably happen congruently with the next chip port. We have enough experience porting to different boards that we could probably write some documentation for that fairly easily, but porting chips has still been pretty iterative. Luckily, I'm nearly always on the IRC channel and happy to help guide folks along!

yexiaobo-seeedstudio, 2017-03-20:
@alevy Cool, and what chip you are porting? Documents is very usefully to help us to porting other platforms.

BTW, I think the board configure file need to separate from build system as an independent "mcu/cpu configure file", such as "mcuconf". It should be more clearly...

kfihihc, 2017-03-20:
@alevy Cool, and what chip you are porting? Documents is very usefully to help us to porting other platforms.

BTW, I think the board configure file need to separate from build system as an independent "mcu/cpu configure file", such as "mcuconf". It should be more clearly...

alevy, 2017-03-20:
@kfihihc I don't think there is an ongoing effort yet to port an STM32 chip, just some interest. We are currently using the Atmel SAM4L and the Nordic NRF51.

I'm not sure what you mean by your second point... we have separate crates (Rust packages) for MCUs (in the `chips` subdirectory) and boards, where the board package imports the MCU package. Is that what you mean?

alevy, 2017-03-24:
@kfihihc & @stephanbuys, if you're interested in helping port Tock to an STM32, I encourage you to joining the [mailing list](https://groups.google.com/forum/#!members/tock-dev) if you haven't already. I have an STM32F3Discovery so can help. Looking through @japaric's excellent book, it seems like getting support for at least the GPIOs/LEDs, UART and I2C should be reasonably straight forward.

I'm closing this issue, but feel free to keep discussing here or on the mailing list.

closed 2017-03-24 by alevy