tock/tock #320: App hangs indefinitely after Signpost API call

adkinsjd, 2017-03-22:
This code https://github.com/lab11/signpost/blob/5ac88756cb14a02e3599e35fc87e4c5761ed823c/software/apps/audio_module/sense_spectrum_send_bytes/main.c eventually will hang indefinitely (after about 1-20 minutes). It always hangs after the signpost_networking_send_bytes call.

Neither the app watchdog, nor the kernel watchdog seem to trigger and restart the app. I have only been able to fix it by resetting the module through its reset button, or through a power cycle.

frenicth, 2017-03-23:
Ran in to the same problem (probably).

Steps to reproduce:
Call delay_ms in userland
Any system call that gets called after this that uses set_alarm more than once fails to return to the user land application.

https://github.com/frenicth/tock/blob/nrf51/ble_radio_virtual_alarm/userland/examples/radio_nrf51dk/main.c#L34-L38

First timer:
https://github.com/frenicth/tock/blob/nrf51/ble_radio_virtual_alarm/capsules/src/radio_nrf51dk.rs#L184-L187

Should be a "periodic" call. 
https://github.com/frenicth/tock/blob/nrf51/ble_radio_virtual_alarm/capsules/src/radio_nrf51dk.rs#L92-L115

How ever, the system calls are not returning in userland

alevy, 2017-03-24:
@frenicth in your case, the problem is actually a bug in the RTC controller implementation in `chips/nrf51/rtc.rs`. It turns out that `handle_interrupt` isn't actually _clearing_ the interrupt, just prevent the controller from delivering more interrupts to the PPI.

As a result, the kernel is basically continuously calling that handler, which has priority over the process (since it's a kernel driver). The `command` is actually returning, but the process simply doesn't get scheduled, and in fact, the kernel keeps churning (sending advertisements and everything).

The reason this isn't a problem with delay _only_ (or with a non-periodic kernel timer for that matter) is because the virtual timer actually disables the underlying hardware timer if there are no outstanding timer events.

The fix is pretty simple, just add to `interrupt_handler` in `chips/nrf51/rtc.rs`:

```rust
        rtc1().events_compare[0].set(0);
```

so the result function is:

```rust
    pub fn handle_interrupt(&self) {
        rtc1().intenclr.set(COMPARE0_EVENT);
        rtc1().events_compare[0].set(0);
        self.callback.get().map(|cb| { cb.fired(); });
    }
```

There should probabaly be a bit of a pass on that driver since it's really not exposing much of the functionality, but this should do it for now (I tested and it works).

@adkinsjd still looking into yours.

alevy, 2017-03-24:
@adkinsjd I _think_ I'm able to replicate the issue you're seeing. For me, it blocks somewhere at the top of the loop (not sure exactly where yet), as opposed to _in_ the watchdog_tickle or signpost_send calls. Is that right?

alevy, 2017-03-24:
Update, @adkinsjd's app _seems_ to be working if I bump the `delay_ms` calls from 1ms to a bit higher (tested with 5ms and 10ms so far). I'm going to leave it running for a while then test again with 2ms to be a bit more confident that's real evidence.

If this is the case, it's (a) unrelated to @frenicth's issue, which I believe is a bug in the NRF51 RTC code and (b) is related to an old bug I thought I had resolved in the virtual_timer that it basically _may_ skip very short delays because by the time the alarm is set, the timer could have already expired.

If this ends up being the case, it should just be a matter of getting back into the nitty gritty of the virtual_timer.

adkinsjd, 2017-03-25:
That does sound correct.

alevy, 2017-03-25:
OK, I _believe_ I found what seems to be a "bug" in the SAM4L's AST driver (so in some sense it _is_ similar to the bug @frenicth saw!)

The SAM4L's AST `set_alarm` looks like this:

```rust
fn set_alarm(&self, tics: u32) {
    self.disable();
    while self.busy() {}
    unsafe {
        (*self.regs).ar0.set(tics);
    }
    self.clear_alarm();
    self.enable_alarm_irq();
    self.enable();
}
```

This is:

  1. Disabling the AST
  2. Setting the alarm0 register
  3. Clearing the ALARM0 interrupt bit
  4. Ensuring the ALARMAST IRQ is enabled
  5. Re-enabling the AST

When I originally implemented this, steps 1 and 5 made sense because that lets us not think about whether the the current clock values exceeds the new alarm value before we set it. Sure it's technically injecting a pause in the clock, but how long could this possibly be?

Turns out a while... The AST has a busy bit that can be checked to see when the hardware is done processing changes, e.g. after re-enabling it. Adding a `let i mut = 0; while self.busy { i += 1 }` to the end of this function and outputing the value of `i` when it completes tells us that it takes ~345 iterations waiting on busy, which is roughly 30uS at 48Mhz (the loop test + body is about 5 instructions).

Since we've already spent a bunch of time in the virtualization layers, 30uS (x2 probably: once for disable, once for enable) is nothing to scoff at. For reference, it looked like by the time virtualization was all said and done, we're left with about 3 or 4 clock tics at 16kHz until our target alarm for 1s (so around 200uS).

All of this basically raises the likelihood we'll miss an already very short alarm. I'm not really sure still what's actually going on with the AST while it's re-enabling... is it possible that the counter gets re-enabled before the interrupt does and in some corner cases we actually miss the alarm even though the counter technically passed over it? I believe our interrupt handing is robust to the other possibilities I've thought through, but I'm not sure.

Either way, both adding any sort of `while self.busy {}` at the end of that function, as well and removing  the basically unnecessary `disable`/`enable` calls seems to at least significantly extend the life of @adkinsjd's app. So I think we're getting warmer. I'm letting it run and seeing if it borks again.

alevy, 2017-03-25:
Everything I've tried so far _eventually_ blocks...

Upon further debugging, indeed, the bug seems to come up specifically when the alarm is set to `<=` 4 tics away from the current counter value, which is just around 250uS. In GDB, I was able to confirm that once blocked, the alarm is properly set up with an alarm value that's in the past but the interrupt is not asserted.

There is also this somewhat troubling language in the datasheet (Section 19.5.3.2):

> Because of synchronization, the transfer of the alarm value will not happen immediately. When
changing/setting the alarm value, the user must make sure that the counter will not count the
selected alarm value before the value is transferred to the register. In that case, the first alarm
interrupt after the change will not be triggered.

But nothing I can find for _how_ to ensure the counter will not count the selected alarm value... like... how many cycles does it take?

alevy, 2017-03-26:
PR available #333.

Seems like always setting the alarm to a minimum of at least 8 tics from the current value works and should be pretty safe.

phil-levis, 2017-03-31:
On a plane so can’t see the PR, but it would be super useful to add a detailed comment for why this is done, referencing the data sheet.

Phil

alevy, 2017-03-31:
@phil-levis I think the answer is we don't totally know because the datasheet is ambiguous (as noted and cited up-thread). When you have a chance to look at the thread more carefully and the code committed in the PR, let us know what other kind of detail would be useful to comment.

closed 2017-03-31 by alevy

phil-levis, 2017-03-31:
Comment looks perfect!