tock/tock #326: Lingering Make Cleanups

ppannuto, 2017-03-24:
Stuff noted while reviewing #325 that's orthogonal to #325

 - [ ] https://github.com/helena-project/tock/pull/325/files#diff-28dd6258eca7e18969c0c42efa4ee139R8 is ugly. Don't check if we're in userland and bail, just don't have a file name `Makefile` (signpost does this right with `AppMakefile.mk` as the common root)
 - [ ] Same with libtock Makefile
 - [ ] Fixup BUILD*DIR family of variable names
 - [ ] Pretty sure there's a way to consolidate the .cc and .cpp recipes 
 - [ ] Generally document all the make variables - mostly the stuff now in Configuration.mk (thanks Branden!)

alevy, 2017-03-24:
`libtock` should be buildable independently, though. We _want_ to let someone just build `libtock.a`

brghena, 2017-03-24:
That's not too hard to do. When I was changing how it built in #325 I had it building on its own. The trick is that it relies on the Configurations.mk and Helpers.mk files, but I don't know how to do the equivalent of a C Header Guard to stop them from being included twice.

alevy, 2017-03-24:
Will it break if it gets included twice?

alevy, 2017-03-24:
also, could just literally run `make` from libtock as part of the app build process

ppannuto, 2017-03-24:
Include guards are pretty much the same in Make and C:

```make
ifndef TECHNICALLY_AN_ENVIRONMENT_VARIABLE_BUT_WHO_CARES
TECHNICALLY_AN_ENVIRONMENT_VARIABLE_BUT_WHO_CARES=1

...
endif
```

While I _think_ the stuff on common/helpers should be fine if included multiple times, no reason to further overcomplicate our lives

ppannuto, 2017-03-24:
As for the running make, I had such hope for https://www.microsoft.com/en-us/research/wp-content/uploads/2016/03/hadrian.pdf when I saw a 2016 paper with that title, but no.

It remains the case that if you want to get dependency checking correct in make you have to figure out how to get it all to happen in one `make` process (or not use `make`... man can dream).

Should still be do-able to make libtock build independently however. And some of the macro tricks that the old GHC make system used look useful (`$$$$` here we come!)

bradjc, 2017-03-25:
- [ ] Move Makefile.common to `/boards`

brghena, 2017-03-26:
#325 should handle everything in this issue except consolidating the cc and cpp recipes

ppannuto, 2017-06-24:
Upon some reading, some futzing, and some failures, I don't think there's actually a good / robust way to combine the .cc and .cpp rules. I pulled the ol' copy-paste trigger again to allow .cxx for C++ files as well in f9f8f4d7 and with that I think we can consider this closed.

closed 2017-06-24 by ppannuto