tock/tock #327: Add LLVM backend to userland Travis

ppannuto, 2017-03-24:
With the release of LLVM 4.0, the requisite PIC support should be around (and will eventually lead to Rust applications when Rust updates LLVM).

In the meantime, it'd be good to prove that the LLVM fixes work for our current C applications so we know if there are unexpected barriers lurking for eventual Rust userland apps

niklasad1, 2017-04-28:
It seems like it is supported now in the nightly channel and will ship in Rust 1.19 according the guys on the channel #rust at IRC but maybe you already knew that, see below:

<mbrubeck> niklasad1: It looks like the LLVM 4.0 upgrade just landed in nightly four days ago, so I think it'll 
<niklasad1> @rkruppe-phone, @mbrubeck ok nice, but I'm interested in PIC support for ARM
<mbrubeck> niklasad1: okay, so it should ship in Rust 1.19 (currently on nightly)

alevy, 2017-04-28:
LLVM _does_ support it, although in my early tests I've been able to generate correct looking object files, but haven't had success linking them if they have relocations (I think some discrepency between the way LLVM generates relocations and how GNU `ld` reads them or something).

I have a [PR out](https://github.com/rust-lang/rust/pull/41560) in Rust that I expect to be merged soon that surfaces these in Rust as well, which should totally mean we could have Rust binaries in userland!!

alevy, 2017-04-29:
Experimented a bit with this yesterday just using C. It's tricky (or maybe impossible) to link object files that use clang's `-fropi -frwpi` using GNU LD since it actually uses a slightly different kind of relocation than GCC does (at least using the flags we use), however, it's definitely possible to generate a correct assembly file that can be assembled with GNU as (at least for very very basic examples).

I expect the same to extend to Rust, but haven't verified yet. I also haven't tested more "interesting" schemes requiring relocations as we did with GCC.

At a high level, the difference with LLVM's ROPI/RWPI from GCC's `-msingle-pic-base` is that LLVM generates references to data that are directly relative to the base of memory (rather than doing an indirection through a GOT). References to text are all relative (not yet sure what that means for references from the data segment to the text segment). This makes sense for most vanilla global variables, but I'm not yet sure I understand how/whether this plays well with nested data references or references to the text segment from global variables that would need to be fixed up.

alevy, 2017-07-26:
This is blocking on LLD 5.0 being released

ppannuto, 2017-07-26:
This is the tracking bug for LLVM/LLD 5.0
https://bugs.llvm.org/show_bug.cgi?id=33849

lygstate, 2018-03-30:
Are this done?

niklasad1, 2018-03-30:
Yes it is merged, but is probably better to use LLVM 6.0 and LLD 6.0.0.

Rust is also using LLVM 6.0 https://github.com/rust-lang/rust/pull/47828

lygstate, 2018-03-30:
I means, are we could be able use rust to building userland apps or libs for TockOS?

alevy, 2018-03-30:
@lygstate yes: https://github.com/tock/libtock-rs

bradjc, 2018-05-22:
Given that we are using rust apps, it doesn't seem like anyone is going to make this happen for our c apps.

closed 2018-05-22 by bradjc