tock/tock #329: Hail and Blink apps don't fit on a Hail

bradjc, 2017-03-24:
Trying to program both the `tests/hail` and `blink` apps on a hail I get:

```
---| App Status |---
TOCK_DEBUG(0): src/main.rs:388: Initialization complete. Enterin

Kernel panic at /home/bradjc/git/tock/kernel/src/process.rs:1095:
	""blink" failed to load. Stack + Data + GOT + BSS (2064) > available memory (0)"
```

`tockloader list --verbose`:

```
[App 0]
  Name:                  hail
  Total Size in Flash:   65536 bytes
  Flash Start Address:   0x00030000
  Flash End Address:     0x0003ffff
  Entry Address:         0x000300ad
  Relocate Data Address: 0x0003004c (length: 96 bytes)
  Text Address:          0x000300ac (length: 36706 bytes)
  GOT Address:           0x0003900e (length: 680 bytes)
  Data Address:          0x000392b6 (length: 157 bytes)
  Minimum Stack Size:    2048 bytes
  Minimum Heap Size:     2048 bytes
  Minimum Grant Size:    1024 bytes
  Checksum:              0x0001172c

[App 1]
  Name:                  blink
  Total Size in Flash:   512 bytes
  Flash Start Address:   0x00040000
  Flash End Address:     0x000401ff
  Entry Address:         0x0004004d
  Relocate Data Address: 0x0004004c (length: 0 bytes)
  Text Address:          0x0004004c (length: 208 bytes)
  GOT Address:           0x0004011c (length: 16 bytes)
  Data Address:          0x0004012c (length: 0 bytes)
  Minimum Stack Size:    2048 bytes
  Minimum Heap Size:     1024 bytes
  Minimum Grant Size:    1024 bytes
  Checksum:              0x00000b85
```

Should this work? It seems like those two apps should fit.

brghena, 2017-03-24:
@ppannuto It looks to me like they should both round up to 8 kB of RAM, which ought to fit in the 16 kB Hail has reserved for apps...

alevy, 2017-03-24:
the `hail` app is:

```bash
   text    data     bss     dec     hex filename
  36702     837    8432   45971    b393 build/hail/app.elf
```

So should round up to 16KB of RAM

alevy, 2017-03-24:
@bradjc does that resolve this question?

bradjc, 2017-03-25:
Well the next question is why aren't we using more of the SAM4L RAM? Some super quick testing seems to indicate that 32k RAM for apps and four slots works.

alevy, 2017-03-25:
No good reason. We should totally bump this to basically however much fits.

closed 2017-03-25 by bradjc