tock/tock #331: Firestorm dev board links are dead

ruabmbua, 2017-03-25:
I just noticed, that both links at _tock/boards/storm/README.md_, which point to the Firestorm dev board appear to be dead. Not sure if it is a temporal issue, but the entire website for the Software defined Buildings Research Group is offline.

A while ago the dev board caught my interest, but now the corresponding websites are offline. If anyone has any info on this, it would be great.

Also if the Firestorm project is actually dead, then the hyperlinks should be removed.

I will revisit the Firestorm related websites a few days from now, and when the server comes back online also delete this issue.

alevy, 2017-03-25:
Thanks for pointing this out @ruabmbua. Storm is the the original platform we built Tock for and it has been more and more abandoned by the original creator of the board in the last couple years. I think it makes sense for us to remove it from Tock anyway, but I'll e-mail the author of the board and ask if the site being down signals an official abandonment.

ruabmbua, 2017-03-26:
Okay thanks for the info. I then will look into some other 1st class supported development boards.

ppannuto, 2017-04-06:
Fixed by #349

closed 2017-04-06 by ppannuto

alevy, 2017-04-07:
@ruabmbua n.b. the new default board is obtainable thanks to the kind efforts and investment of @bradjc @ppannuto and @brghena <https://www.tockos.org/hardware/hail>