tock/tock #346: Consolidate PLATFORM and TOCK_BOARD variables

ppannuto, 2017-04-02:
Looks like the kernel build uses PLATFORM and the user build uses TOCK_BOARD to refer to the same thing. We should pick one and use consistently

bradjc, 2017-05-26:
Probably the best way to handle this is to wait until all platforms support tockloader and then we won't need `TOCK_BOARD` in userland.

alevy, 2017-07-26:
Blocked on "all platforms support[ing] tockloader"

ppannuto, 2018-05-08:
Update as of May 8, 2018:

[Tockloader supports](https://github.com/tock/tockloader/blob/master/tockloader/board_interface.py#L9):

  - hail
  - imix
  - nrf51
  - nrf52

In addition to those boards, Tock mainline currently has:

  - ek-tm4c1294xl [Tracking Issue](https://github.com/tock/tockloader/issues/25)
  - launchxl [Tracking Issue](https://github.com/tock/tockloader/issues/26)