tock/tock #351: SAM4L I2C does not sense channel

ppannuto, 2017-04-06:
Okay.. so this is a problem. I'm hoping it's something that we're just not configuring correctly, but frankly I'm shocked this is even a thing the I2C controller would ever let happen.

The setup:
  - Two nodes sending shortish (~50 byte) I2C messages in a tight loop to a third node

Tock timing modification:
  - Toggle a GPIO at the beginning and end of an I2C master write
```
$ git diff
diff --git a/chips/sam4l/src/i2c.rs b/chips/sam4l/src/i2c.rs
index 6ee3f179..218a582a 100644
--- a/chips/sam4l/src/i2c.rs
+++ b/chips/sam4l/src/i2c.rs
@@ -21,6 +21,8 @@ use kernel::hil;
 use nvic;
 use pm;

+use gpio;
+
 // Listing of all registers related to the TWIM peripheral.
 // Section 27.9 of the datasheet
 #[repr(C, packed)]
@@ -273,6 +275,7 @@ impl I2CHw {
                             }
                             None => None,
                         };
+                        unsafe { gpio::PA[06].toggle(); }
                         buf.map(|buf| { client.command_complete(buf, err); });
                     });
                 });
@@ -385,6 +388,7 @@ impl I2CHw {
             dma.prepare_xfer(self.dma_pids.1, data, len as usize);
             self.setup_xfer(chip, flags, false, len);
             self.master_enable();
+            unsafe { gpio::PA[06].toggle(); }
             dma.start_xfer();
         });
     }
```

This is the start of a normal transaction here:
[image]

Notice the 170 ns gap between when the I2C start condition comes and our GPIO toggles.

Now if we move forward to later in this transaction we'll see this:

[image]

Key things to see:

 - After sending 0x28 (correctly), the first node is trying to send 0x31 here (Saleae decodes it as 0x28, but that's not what's happening)
 - The GPIO timing toggle for the second node goes low, it is trying to send
 - 170 ns before that, we see SDA go low
    - interestingly, this is also exactly when node 1 was driving clock low, which is too weird to be a coincidence
 - At this point, node 2 is trying to send a write to address 0x20, note this is `010 0000 0` as I2C addresses are 7 bits + R/W
 - The long held low clock isn't a problem on its own, however when the clock line does go high again (the first rising edge after the timing marker), node 1 (the original master) was trying to send a high bit (the 0x31 --> 001*1* 0001), however node 2 is trying to send a 0 (*0*10 0000 0).
 - Node 1 correctly observes this and backs off the rest of the transaction
 - Node 2 believes it owns the bus, so it dutifully continues clocking, the next edge is a 1 as expected, followed by all zeros
 - It's quite probable that on the sixth clock edge "driven" by node 2, the receiving node ACK'd (drove SDA low), but this is obscured by node 2 driving a data 0 here [Saleae identifies the ACK/NAK edge by not drawing a rising arrow. Saleae draws no arrows on the remaining edges afterwards because they do not encode a complete byte]
 - When node 2 gets around to clocking the ACK/NAK cycle for its "first byte", the receiving node thinks it's in the middle of getting the next byte, so it naturally does not ACK
 - Since the destination address byte was NAK'd, Node 2 gives up and issues a stop

End result:
  - Node 1 reports a multi-master loss to the upper levels
  - Node 2 reports a NAK no one listening to the upper levels
  - And the receiver gets cut off in the middle of a byte

I'm _hoping_ that there's something in the bowels of the SAM4L datasheet that will convince the chip not to blindly write when it's the master, but I don't think this is a problem that will be solved at 3AM

ppannuto, 2017-04-06:
To replicate

  - Grab the `i2c-trample` branch. Since tends to occur towards the end of messages, it rips out the crypto and overwrites the hmac/hash field with a predictable, known byte pattern
  - Flash two modules with `tests/i2c_blast_packets` - mind the `#define MODULE_ID` is different
  - Flash a controller with `controller/signpost_controller_enable_init_only`

And you're off to the races. Let it run for 10 or 20 seconds and this'll happen. Easy enough to skim the Saleae capture to find transactions that look too short

ppannuto, 2017-04-06:
Okay I wanted to sleep but I couldn't not read a little. The SAM4L is _supposed_ to take care of this:

[image]
^p716

brghena, 2017-04-06:
Having similarly decided to do some research, the TWI multi-master mode has been know to be buggy and bad on AVRs for more than a decade. I can't promise the same hardware problem is affecting the SAM4L, but it sure seems probable, doesn't it?

http://www.robotroom.com/Atmel-AVR-TWI-I2C-Multi-Master-Problem.html
http://www.avrfreaks.net/comment/211004#comment-211004

ppannuto, 2018-05-08:
I think over time we've convinced ourselves that this is a fault of the SAM4L hardware, so there's not so much that we can do. Would love if the world developed a workaround, but for now I think we just have to close this.

closed 2018-05-08 by ppannuto