tock/tock #353: Tock error numbers and names conflict with errno

ppannuto, 2017-04-08:
```
In file included from /Volumes/code/lab11/signpost/software/kernel/tock/userland/libtock/led.h:3:0,
                 from main.cc:11:
/Volumes/code/lab11/signpost/software/kernel/tock/userland/libtock/tock.h:30:0: warning: "EBUSY" redefined
 #define EBUSY    -2

In file included from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/errno.h:9:0,
                 from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/c++/6.3.1/cerrno:42,
                 from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/c++/6.3.1/ext/string_conversions.h:44,
                 from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5417,
                 from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/c++/6.3.1/string:52,
                 from /usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/c++/6.3.1/random:40,
                 from main.cc:9:
/usr/local/Cellar/arm-none-eabi-gcc/6-2017-q1-update/arm-none-eabi/include/sys/errno.h:48:0: note: this is the location of the previous definition
 #define EBUSY 16 /* Device or resource busy */
```

The way I see it we can either remap our numbers to match the name/number from errno, or we can put our errors into a namespace (i.e. `TOCK_EBUSY`). While our definitions are similar to errno, I think I still lean towards the private namespace so as not to imply an exact match or tie ourselves to errno for the future.

Thoughts?

phil-levis, 2017-04-08:
I agree that TOCK_EBUSY is better — it definitively solves the issue. But why are we pulling in errno.h? 

Phil

ppannuto, 2017-04-08:
In this case it's getting pulled in on my behalf. I wanted to run a quick experiment and (bless the trivial availability of C++ on an MCU in userland Tock) it was a better experiment to send things based on a Poisson distribution, which is [part of the standard library as of C++11](http://www.cplusplus.com/reference/random/poisson_distribution/)

When I `#include`d random, it ended up chaining back to errno (from that output looks like random->string->basic_string->string_conversion->cerrno->errno)

PR #461 (2017-06-26): userland: prepend `TOCK_` to error names
This allows `tock.h` to be compatible with `errno.h`.

Fixes #353
commit b1d5842c (2017-06-26): userland: prepend `TOCK_` to error names

This allows `tock.h` to be compatible with `errno.h`.

Fixes #353
closed 2017-06-26 by ppannuto