tock/tock #379: Feature Idea: Enable/disable applications & "permanent" applications

bradjc, 2017-05-17:
### Idea

Extend the TBF header to add two bits:

1. Enable/disable flag. An app that is enabled runs on boot, just like apps do now. If an app is disabled, the kernel does not load and run it.
2. Permanent flag. An app that is permanent will not be removed by `tockloader erase-apps` unless an extra flag is added to the command line.

### Why?

I'm imagining that we could add test apps or admin apps or something to the default hail/imix image that would just sit dormant normally. Then we could have people run a tockloader command that would activate the apps to help debug.

The enable/disable flag would also be useful for misbehaving apps as the kernel would have a method for marking that it shouldn't boot the app.

Permanent apps might also be useful for "library" applications that are stable. Then the user could work on other apps without having to worry about the library app being removed.

ppannuto, 2017-05-17:
I think I like the concept, but there's some UX in tockloader that'll be
important to not make this obnoxious, things that come to mind:

1. After running erase-apps, tockloader must print something like:
  Erased:
     accel-leds
     blink
  Did Not Erase Sticky Apps
     vendor-info
     hail-debug [Disable]
  To erase these apps as well, run `tockloader erase-apps --all`

Key things:
  - explains what was and was not deleted
  - explains why things weren't deleted
  - explains how to do what you might've wanted instead

2. I think an "app summary" should print after every tockloader app-related
operation

   Installed Apps:
      Active
         accel-leds
      Disabled
         hail-debug
    Use `tockloader [enable|disable] app` to turn apps on or off

ppannuto, 2017-05-17:
Your use of the word "library" applications is also interesting, and may
expose a third class of apps we should consider.

I'm imagining the user-space networking library that's running primarily as
an IPC service. It's not a vendor app (i.e. `hail-debug`). Perhaps we
should have a `library` or `service` bit for "apps" that are exclusively
services to other apps

commit a68e51a7 (2017-05-28): kernel: Update to TBF Header v2

Tock Binary Format header version 2 adds a `flags` field that can be
used to mark properties of a flashed application.

The two supported flags are:

1. Enable/disable. Tells the kernel to run or not run an application at
boot.
2. Sticky. Tells tockloader to not erase the app without --force.

Fixes #379
commit 46642361 (2017-07-20): kernel: Update to TBF Header v2

Tock Binary Format header version 2 adds a `flags` field that can be
used to mark properties of a flashed application.

The two supported flags are:

1. Enable/disable. Tells the kernel to run or not run an application at
boot.
2. Sticky. Tells tockloader to not erase the app without --force.

It also allows apps to tell the kernel to not do PIC fixups on its
behalf.

Apps can also now specify flash regions that the app intends to write.

Fixes #379
commit f8242a60 (2017-07-20): kernel: Update to TBF Header v2

Tock Binary Format header version 2 adds a `flags` field that can be
used to mark properties of a flashed application.

The two supported flags are:

1. Enable/disable. Tells the kernel to run or not run an application at
boot.
2. Sticky. Tells tockloader to not erase the app without --force.

It also allows apps to tell the kernel to not do PIC fixups on its
behalf.

Apps can also now specify flash regions that the app intends to write.

Fixes #379
commit 955aed6a (2017-07-20): kernel: Update to TBF Header v2

Tock Binary Format header version 2 adds a `flags` field that can be
used to mark properties of a flashed application.

The two supported flags are:

1. Enable/disable. Tells the kernel to run or not run an application at
boot.
2. Sticky. Tells tockloader to not erase the app without --force.

It also allows apps to tell the kernel to not do PIC fixups on its
behalf.

Apps can also now specify flash regions that the app intends to write.

Fixes #379
closed 2017-07-22 by bradjc