tock/tock #382: Possible MPU problem

brghena, 2017-05-20:
I'm having a problem running the `userland/examples/tests/adc_continuous` app from #371. Everything was working fine before I rebased on top of master.

If I'm reading the debug report below correctly, it's erroring on address `0x20008E10` which is validly within the Data segment.

```
Kernel panic at /home/brghena/Dropbox/repos/tock/kernel/src/process.rs:295:
    "Process adc_continuous had a fault"
    Kernel version b378e87

---| Fault Status |---
Data Access Violation:              true
Forced Hard Fault:                  true
Faulting Memory Address:            0x20008E10
Fault Status Register (CFSR):       0x00000082
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: adc_continuous   -   [Fault]
 Events Queued: 0   Syscall Count: 0   Last Syscall: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x2000C000═╪══════════════════════════════════════════╝
             │ ▼ Grant         268 |   1024          
  0x2000BEF4 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ Unused
  0x20008E20 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ ▲ Heap            0 |   1024              S
  0x20008E20 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ R
             │ Data          17952 |  17952              A
  0x20004800 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ M
             │ ▼ Stack         120 |   2048          
  0x20004788 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00032000 ┬───────────────────────────────────────────
             │ Unused
  0x00031A11 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ F
             │ Data            298                       L
  0x000318E7 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ A
             │ Text           6259                       S
  0x00030074 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ H
             │ Header          116
  0x00030000 ┴───────────────────────────────────────────

  R0 : 0x200048B4    R6 : 0x200048B4
  R1 : 0x00000010    R7 : 0x20004800
  R2 : 0x20008E10    R8 : 0x000317B7
  R3 : 0x000000A4    R10: 0x00000000
  R4 : 0x00000000    R11: 0x00000000
  R5 : 0x00000018    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x20004788 (Process Stack Pointer)
  LR : 0x00030537 [0x800004C2 in lst file]
  PC : 0x0003053A [0x800004C6 in lst file]
 YPC : 0x00030075 [0x8002E802 in lst file]

 APSR: N 0 Z 0 C 0 V 0 Q 0
       GE 0 0 0 0
 IPSR: Exception Type - Thread Mode
 EPSR: ICI.IT 0x00
       ThumbBit true 
```

brghena, 2017-05-20:
From the disassembly, 0x20008e10 should be `__malloc_free_list`, which checks out given that the PC points to somewhere in the `_malloc_r` function. This sure looks like an MPU bug to me.

```
Disassembly of section .bss:

0000011c <_bss>:
        ...

0000011e <sample_buffer1>:
        ...

00002392 <sample_buffer2>:
        ...

00004608 <putstr_head>:
    4608:       0000            movs    r0, r0
        ...

0000460c <putstr_tail>:
    460c:       0000            movs    r0, r0
        ...

00004610 <__malloc_free_list>:
    4610:       0000            movs    r0, r0
        ...

00004614 <__malloc_sbrk_start>:
    4614:       0000            movs    r0, r0
        ...

00004618 <__dso_handle>:
    4618:       0000            movs    r0, r0
        ...

0000461c <errno>:
    461c:       0000            movs    r0, r0
        ...
```

alevy, 2017-05-21:
It's definitely a bug. It's basically the same issue as with misaligned text segments when the length is large. Note that, the size of RAM is `0x8000` (32kB) and the base is `0x20004000`, which is not divisible by 32kB. 

In previous version of master, there was also an off-by-one bug in setting the MPU the sufficed to compensate this, but actually over exposed RAM.

The fix should be similar, basically use subregions in these cases to make a large enough region and split it up. Working on a fix now.

PR #384 (2017-05-23): Generalize MPU subregions and use where possible
E.g. for text sections, data sections, grants and ipc regions.
Unfortunately, this currently makes the MPU hil a bit more Cortex-M MPU
specific... working on it.

This is somewhat in progress, but I think it might be worth merging meanwhile.

At this point, it make sense to makes the `Region` type to an associated of the `MPU` trait so it can be implementation specific and cache `Regions` between calls so the calculation is required each time (should drastically reduce the context switch time). However, both of these introduce to typing issues t [... cut]
closed 2017-05-23 by alevy