tock/tock #390: Cannot build nrf51dk "hex" target

brghena, 2017-05-25:
From @rasit

> I was trying the sample programs on a nrf51dk board, but it seems the instructions below does not match the real thing:
> 
> https://github.com/helena-project/tock/blob/master/boards/nrf51dk/README.md
> 
> when I say make TOCK_BOARD=nrf51dk hex, I get " no rule to make target "hex"
> 
> what am I missing? I just followed the instructions and installed everything on Ubuntu 16.04 (well, the Windows version of it;-)

I'm not very familiar with the `nrf51dk` platform. Anyone else have ideas about this?

rasit, 2017-05-25:
I managed to install the kernel and the blink app onto the board 
successfully via tockloader (on Ubuntu 16.04 vm)

However, i noticed that it is no longer make ... program, but flash;-)

I found a few issues in the documentation/instructions that i'll report 
later.

Cheers,
--Rasit

<mention@noreply.github.com>
Sent: 2017-05-25 12:12:21 PM
Subject: [helena-project/tock] Cannot build nrf51dk "hex" target (#390)

alevy, 2017-05-25:
@rasit since you're already have the nrf51dk working, you may want to check out @niklasad1 and @frenicth's ongoing work on adding BLE advertising support to the kernel (it's working just needs to be cleaned up, and merged): https://github.com/frenicth/tock/tree/nrf51/ble_radio

rasit, 2017-05-25:
Will do, thanks! 

⁣Please pardon any typos and erroneous auto-corrections.

Sent from BlueMail ​

niklasad1, 2017-07-25:
Yes, I have verified on nRF51-DK and ```TOCK_BOARD=nrf51dk hex``` is not supported.

As fair as I can tell this is not supported by the [Makefile](https://github.com/helena-project/tock/blob/master/boards/nrf51dk/Makefile).

I would suggest to remove the entire paragraph in [README.md#programming-with-mbed-file-system-currently-unsupported](https://github.com/helena-project/tock/blob/master/boards/nrf51dk/README.md#programming-with-mbed-file-system-currently-unsupported)

The path ```boards/nrf51dk/target/nrf51/release/nrf51dk.hex``` is also wrong since we emerged to use the xargo linker it is now ```boards/nrf51dk/target/thumbv6m-none-eabi/release/nrf51dk.hex```

I suppose the way go about this is (not checked though)
```
$ make TOCK_BOARD=nrf51dk
$ cd boards/nrf51dk/target/thumbv6m-none-eabi/release/
$ cp nrf52dk.hex firmware.hex
# move to device!!!
```
However, I have never used this I will probably not do easier to use
```make TOCK_BOARD=nrf51dk flash```

Thoughts?

alevy, 2017-07-25:
I'm for it. It relatively straightforward to convert a .bin to .hex manually if the need arises and it's not longer used in anyway for the normal programming path

ppannuto, 2017-07-25:
+1 for removing it, we were actually just chatting about this yesterday in
lab

closed 2017-07-25 by alevy