tock/tock #414: Possible issue with `putnstr_async`

bradjc, 2017-06-01:
This app doesn't print anything:

```c
#include <stdio.h>
#include <tock.h>
#include <console.h>

void cb (int a, int b, int c, void* d) {}

int main () {
  putnstr_async("test\n", 5, cb, NULL);
  yield();
  return 0;
}
```
This one prints both statements:
```c
#include <stdio.h>
#include <tock.h>
#include <console.h>

void cb (int a, int b, int c, void* d) {
  printf("ok\n");
}

char c[] = "test\n";

int main () {
  putnstr_async(c, 5, cb, NULL);
  yield();
  return 0;
}
```
Is this normal or expected? Also thanks to @adkinsjd for bringing this up.

ppannuto, 2017-06-01:
This is a real bug, inside putnstr_async we aren't checking the return value of `allow` or `subscribe`. In the first instance, allow returned `EINVAL`, likely because that the compiler did something tricky with the fact that the memory for `"test\n"` is technically not valid after that function call (so using the async version this way is also a bug)

alevy, 2017-06-01:
Yes, this is expected. We don't (yet) know how to pass a read-only buffer to the kernel, so `allow` must be passed a memory allocated string rather than a flash-allocated one. If you print the return value from `putnstr_async` it should be -1.

Standard library variants like `printf` and `puts` end up calling `putnstr` (the non-async version), which `malloc`s an empty buffer and copies the passed in string into it, while the `_async` version does not (it assumes it's already been malloced.

alevy, 2017-06-01:
Oh @ppannuto is right, we're not returning early from `putnstr_async` if `allow` fails...

ppannuto, 2017-06-01:
I'll fix the userland library for now

alevy, 2017-06-04:
Can we close this issue, since it's (sort of) expected behavior and @ppannuto fixed the return value issue in #415?

ppannuto, 2017-06-04:
Opened #418 track the underlying issue

closed 2017-06-04 by ppannuto