tock/tock #418: Expand allow interface with permission masks?

ppannuto, 2017-06-04:
i.e. allow read-only, allow write-only

could also imagine 'allow until next subscribe complete' and variations on that theme, but I think the RO vs WO vs RW is the most important

alevy, 2017-07-26:
Any suggestions on what the system call interface would look like?

On the kernel side, we could add a `is_writeable()` method to `AppSlice` that the capsule could use to determine if it can write to the buffer (and return an error if it can't and needs to). I don't we have any registers left in the `allow` system call though. Maybe use the bottom bit of the base pointer, since it'll have to be word aligned?

ppannuto, 2017-07-26:
I think we'd be better off harvesting bits from the `size` parameter, taking anywhere from the top nibble to top half-word for a more generic `flags` field. That would let us handle R/W/X bits, as well as leave an avenue open to the more complex suggestions like 'allow until next subscribe complete', which feels useful for one-shot events.

It'd certainly be valid (IMO) although a touch odd to allow 1 byte that's not aligned. Is there a reason to require word alignment?

alevy, 2017-07-26:
I think non-word alignment might be an issue for architectures that can't do unaligned reads.

However, taking from the size parameter makes sense. 64KB seems like plenty for an individual AppSlice, but let's maybe start with the top byte, and we can expand from there if we actually need it

bradjc, 2018-06-11:
Wishlist, moved to wiki

closed 2018-06-11 by bradjc