tock/tock #420: ADC Buffer Ownership Interface

brghena, 2017-06-05:
Per today's meeting:

- let's change the ADC to have a `stop_sampling_highspeed` function that both stops and returns buffers
- `stop_sampling` in highspeed mode would be an error
- `sample_highspeed` would return an error if the ADC still owns buffers
- change ADC trait to be a tree, with ADC owning init and the channel type, and AdcSimple and AdcHighSpeed inheriting from it

phil-levis, 2017-06-05:
I think it should be OK if we just leave the interface as is for now — we want to figure out what the right general buffer management idioms are, based on USB and 6lowpan as well as ADC? If you want to hack on this, go for it, but we might revisit it in a few weeks?

Phil

alevy, 2017-07-26:
@brghena do we want to leave this open?

brghena, 2017-07-26:
I'm fine with closing it for now. Mostly I just wanted to have the ideas written down so we could come back to them later when more complex drivers (like Phil mentioned) become standardized too.

closed 2017-07-26 by brghena