tock/tock #427: Uncrustify Code formatter seem to not work on Ubuntu

niklasad1, 2017-06-10:
Hi,
I tried to use Uncrustify to test the C formatter in libtock on Ubuntu 17.04 which did not work, see attached prompt log below:

```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.04
Release:	17.04
Codename:	zesty
```
```bash
$ uncrustify -v
uncrustify 0.64
```
```bash
$ make format           
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:87 Unknown symbol 'sp_enum_colon'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:316 Unknown symbol 'sp_after_type_brace_init_lst_open'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:319 Unknown symbol 'sp_before_type_brace_init_lst_close'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:322 Unknown symbol 'sp_inside_type_brace_init_lst'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:335 Unknown symbol 'sp_type_brace_init_lst'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:613 Unknown symbol 'sp_after_newop_paren'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:616 Unknown symbol 'sp_inside_newop_paren'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:619 Unknown symbol 'sp_inside_newop_paren_open'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:622 Unknown symbol 'sp_inside_newop_paren_close'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:637 Unknown symbol 'force_tab_after_define'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:653 Unknown symbol 'indent_param'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:891 Unknown symbol 'indent_using_block'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:897 Unknown symbol 'indent_ternary_operator'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:995 Unknown symbol 'nl_enum_class'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:998 Unknown symbol 'nl_enum_class_identifier'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1001 Unknown symbol 'nl_enum_identifier_colon'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1004 Unknown symbol 'nl_enum_colon_type'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1029 Unknown symbol 'nl_before_if_closing_paren'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1235 Unknown symbol 'nl_type_brace_init_lst'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1238 Unknown symbol 'nl_type_brace_init_lst_open'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1241 Unknown symbol 'nl_type_brace_init_lst_close'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1370 Unknown symbol 'nl_max_blank_in_func'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1590 Unknown symbol 'align_var_def_colon_gap'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1847 Unknown symbol 'mod_full_brace_nl_block_rem_mlcond'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1915 Unknown symbol 'mod_sort_oc_property_class_weight'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1974 Unknown symbol 'pp_ignore_define_body'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1981 Unknown symbol 'include_category_0'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1984 Unknown symbol 'include_category_1'
/home/niklasad1/tock/userland/tools/uncrustify/uncrustify.cfg:1987 Unknown symbol 'include_category_2'
Parsing: main.c as language C
../../../AppMakefile.mk:195: recipe for target 'format' failed
make: *** [format] Error 1
```

ppannuto, 2017-06-10:
Plane just landed and on mobile most of the day, but I'm guessing that's
because some of those options are only for the newest version, 0.65 instead
of 0.64 -- that config file was auto generated by the software with all the
defaults. We should be able to just comment out those unknown options that
we aren't using anyways

brghena, 2017-06-11:
It feels like using `apt` to install uncrustify is problematic. Doesn't that everyone will have different versions of uncrustify depending on which version of ubuntu they are running? Is there a promise that uncrustify is backwards compatible?

ppannuto, 2017-06-12:
It looks like uncrustify wasn't written to be forward-compatible (i.e. ignore unknown config options), which.. meh. For an immediate fix, I just commented out the options that 0.65 added over 0.64. If other versions are actually prevalent in the wild, we can reassess.

niklasad1, 2017-06-12:
Thank you.
I guess specifying the version in [Userland.md](https://github.com/helena-project/tock/blob/master/doc/Userland.md) would be a good start or somewhere more appropriate. 
It should be possible to compile [uncrustify](https://github.com/uncrustify/uncrustify) from source for most operating systems.

bradjc, 2017-06-12:
Yeah doesn't work for me either:

```
./format_all.sh 

Building ./sdcard
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:129 Unknown symbol 'string_replace_tab_chars'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:143 Unknown symbol 'enable_digraphs'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:169 Unknown symbol 'sp_cpp_lambda_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:181 Unknown symbol 'sp_enum_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:199 Unknown symbol 'sp_before_pp_stringify'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:214 Unknown symbol 'sp_cparen_oparen'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:236 Unknown symbol 'sp_after_ptr_star_qualifier'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:242 Unknown symbol 'sp_ptr_star_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:267 Unknown symbol 'sp_before_template_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:286 Unknown symbol 'sp_angle_paren_empty'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:296 Unknown symbol 'sp_permit_cpp11_shift'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:308 Unknown symbol 'sp_inside_sparen_open'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:359 Unknown symbol 'sp_after_mdatype_commas'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:362 Unknown symbol 'sp_before_mdatype_commas'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:365 Unknown symbol 'sp_between_mdatype_commas'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:380 Unknown symbol 'sp_after_constr_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:383 Unknown symbol 'sp_before_constr_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:395 Unknown symbol 'sp_after_operator_sym_empty'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:432 Unknown symbol 'sp_func_proto_paren_empty'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:438 Unknown symbol 'sp_func_def_paren_empty'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:447 Unknown symbol 'sp_inside_tparen'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:450 Unknown symbol 'sp_after_tparen_close'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:459 Unknown symbol 'sp_fparen_dbrace'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:476 Unknown symbol 'sp_func_class_paren_empty'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:491 Unknown symbol 'sp_after_throw'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:506 Unknown symbol 'sp_super_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:509 Unknown symbol 'sp_this_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:545 Unknown symbol 'sp_word_brace'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:548 Unknown symbol 'sp_word_brace_ns'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:598 Unknown symbol 'sp_after_oc_dict_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:602 Unknown symbol 'sp_before_oc_dict_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:644 Unknown symbol 'sp_after_oc_msg_receiver'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:647 Unknown symbol 'sp_after_oc_property'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:653 Unknown symbol 'sp_cond_colon_before'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:656 Unknown symbol 'sp_cond_colon_after'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:662 Unknown symbol 'sp_cond_question_before'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:665 Unknown symbol 'sp_cond_question_after'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:668 Unknown symbol 'sp_cond_ternary_short'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:677 Unknown symbol 'sp_after_for_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:680 Unknown symbol 'sp_before_for_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:683 Unknown symbol 'sp_extern_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:689 Unknown symbol 'sp_cmt_cpp_doxygen'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:692 Unknown symbol 'sp_cmt_cpp_qttr'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:701 Unknown symbol 'sp_between_new_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:710 Unknown symbol 'sp_annotation_paren'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:713 Unknown symbol 'sp_skip_vbrace_tokens'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:763 Unknown symbol 'indent_paren_open_brace'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:766 Unknown symbol 'indent_cs_delegate_brace'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:773 Unknown symbol 'indent_namespace_single_indent'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:793 Unknown symbol 'indent_class_on_colon'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:796 Unknown symbol 'indent_constr_colon'
[... cut ...]
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:2021 Unknown symbol 'use_options_overriding_for_qt_macros'
/home/bradjc/git/tock/userland/tools/uncrustify/uncrustify.cfg:2028 Unknown symbol 'warn_level_tabs_found_in_verbatim_string_literals'
Parsing: main.c as language C
../../AppMakefile.mk:195: recipe for target 'format' failed
make: *** [format] Error 1
 ⤤ Failure formatting ./sdcard
```

```
uncrustify --version
uncrustify 0.59
```

https://packages.ubuntu.com/xenial/uncrustify

brghena, 2017-06-14:
Actually, your fix didn't actually fix anything. I've also got uncrustify 0.59 and cannot format code on master. (Also Rasit is having problems too)

```
Building ./rf233
/home/brghena/workspace/tock/userland/tools/uncrustify/uncrustify.cfg:28 Unknown symbol 'nl_enum_class'
/home/brghena/workspace/tock/userland/tools/uncrustify/uncrustify.cfg:29 Unknown symbol 'nl_enum_class_identifier'
/home/brghena/workspace/tock/userland/tools/uncrustify/uncrustify.cfg:39 Unknown symbol 'nl_max_blank_in_func'
Parsing: main.c as language C
Parsing: rf233.c as language C
../../AppMakefile.mk:195: recipe for target 'format' failed
make: *** [format] Error 1
 ⤤ Failure formatting ./rf233
```

ppannuto, 2017-06-14:
Sigh.. the previous fix tested 0.61, apparently you're even older. As of ea1c664, I've removed those three style rules and added an error that prints if your uncrustify version is older that 0.59

commit 8185e72c (2017-06-12): style: comment out v0.64 options

uncrustify isn't forward compatible, so using a config generated
from bleeding edge version causes some issues

Fixes #427
closed 2017-06-14 by ppannuto