tock/tock #432: Uncrustify still not working

brghena, 2017-06-14:
**Problem 1:** Checking for unstaged changes doesn't work

If I run `format_all.sh` on ubuntu 16.04 (version 0.59), it first checks rf233. It then finds changes to rf233.c and makes them inline. Then it moves to spi_byte, sees that there are unstaged changes to rf233.c and asks if I want to continue. Then it moves to lua-hello, sees that there are unstaged changes to rf233.c and asks if I want to continue...

**Problem 2:** I have changes to rf233.c at all

Since the build is currently passing, does this mean that different versions of uncrustify return different results?

*userland/examples/rf233/rf233.c*
 ```
 enum {
-  RADIO_TX_OK = 0,
-  RADIO_TX_ERR = 1,
-  RADIO_TX_NOACK = 2,
+  RADIO_TX_OK        = 0,
+  RADIO_TX_ERR       = 1,
+  RADIO_TX_NOACK     = 2,
   RADIO_TX_COLLISION = 3
 };
 ```

@ppannuto

bradjc, 2017-06-22:
It would be great to have:

```
$ cd /tock
$ make formatall
```

that runs all of the formatters. This would be useful for getting PRs ready.

brghena, 2017-06-24:
I was thinking about this today, and it seems to me that they way to improve this is to separate `make format` for libraries from `make format` for example code, and have `make formatall` call both. Then developers can either run `make format` in the specific directory they want to handle, or `make formatall` in the root directory to do everything.

We could then create a  `userland/format_libraries.sh` (like `userland/examples/format_all.sh`) which handle the special-purpose logic of which libraries should be formatted with which tools. For now it would just be `libtock` with uncrustify, but you could imagine `lua53` being similarly formatted with its own tool in the future. `formatall` would then just call the library format script and the example app format script.

(Sidebar: I'm posting this here for @ppannuto instead of just doing it because I tried to implement it and failed. I don't even understand what makes the double-colon rule only apply to libtock right now...)

closed 2017-07-06 by ppannuto