tock/tock #437: DAC for SAM4L is not outputting signal

hsiehju, 2017-06-20:
Implementation of the the DAC on the SAM4L

Had problems with setting up the mode register (MR), writing to the conversion data register (CDR), and outputting a signal on the hail board.

MR = 0x00601012
Corresponds with these settings:
-half-word transfer mode 
-start up time (0x60)
-internal trigger (0x10)
-peripheral event
-enable DAC

brghena, 2017-06-20:
The branch with code can be found here: https://github.com/helena-project/tock/tree/dac

alevy, 2017-06-20:
@hsiehju thanks!

bradjc, 2017-06-21:
This was actually a hardware issue. Turns out, the ADVREFP pin needs to be tied to ADDANA to set the range of possible output voltages.

@shaneleonard does imix have ADVREFP tied to something?

closed 2017-06-21 by bradjc