tock/tock #444: HIL: Add generic temperature sensor interface

bradjc, 2017-06-21:
From the mailing list, it sounds like @niklasad1 is looking into this.

alevy, 2017-06-21:
@niklasad1 as you work on this, this is a good place to post things like proposed designs, questions or general progress

niklasad1, 2017-06-21:
Roger that

bradjc, 2017-06-26:
This will also fix #339.

niklasad1, 2017-07-25:
I have started to look at this but the existing "interfaces/capsules" seem to be hardware dependent i.e., device specific constant, enable interrupts and similar.

How do you suggest to approach this?
Create a generic *TemperatureSensor* capsule as a wrapper to existing capsules to begin with?

bradjc, 2017-07-25:
I think the start is defining the HIL layer. Something like what we have for accelerometers: https://github.com/helena-project/tock/blob/master/kernel/src/hil/ninedof.rs

Then all temperature sensors should implement that interface. Something like what the FXO does: https://github.com/helena-project/tock/blob/master/capsules/src/fxos8700cq.rs#L312

Then a single capsule can implement a syscall driver interface for temperature sensors.

The ninedof stuff happens to be a good example of this, but it is also a first attempt, and may not be the best way or the way we want to standardize around in tock. So its a good starting point, but feel free to diverge if it makes sense to.

niklasad1, 2017-07-28:
Yes, your comment helped alot @bradjc 

I started with a very simple version to test with nRF51
[Temperature Capsule](https://github.com/niklasad1/tock/blob/capsule/generic_temp_sensor/capsules/src/temperature.rs)
[Temperature HIL](https://github.com/niklasad1/tock/blob/capsule/generic_temp_sensor/kernel/src/hil/temperature.rs)

I think for external temperature sensors they belong in capsules but it makes not sense for internal temperature sensors e.g., nRF5X which are chip specific but that is no problem only the initialization in boards/xxx/src/main.rs will only differ a little bit. 

What is a proper interface, I guess a sampling method should exist? 
Is ```read_measurement``` ok for a single measurement? 
I have seen you have used ```take_measurement``` is it better?

Thoughts? 

Further, I will start to remove the syscall implementation from [tmp006.rs](https://github.com/niklasad1/tock/blob/capsule/generic_temp_sensor/capsules/src/tmp006.rs) et al but I can't test it thus I need help with that!

niklasad1, 2017-07-28:
Updated a bit, this is my template "solution" and I need some feedback before doing anything more.

Also, regarding libtock an ideal solution would be to have the same syscall number for the temperature sensors in libtock but then we have to assume that every board can only have one temperature sensor active which is probably not a good assumption or maybe use a subcommand number for the different sensors.

Thus, should the the library in libtock have a separate function for each temperature sensor for example: 
- ```read_temperature_nrf()```
- ```read_temperature_tmp006()```
- ```read_temperature_si7021()```

Thoughts?

alevy, 2017-07-28:
I'm still planning on reading through and giving feedback on the design in general, but for the `libtock` question, I think we _definitely_ want a single set of library functions regardless of the specific sensor. That's kind of the point :) So I think just `read_temperature` for all cases. The board configuration has to decide which sensor to use for that. If we want to be _more_ explicit, maybe we should call it `read_ambient_temperature`.

I agree that it's problematic for dealing with boards with multiple temperature sensors, but that's a separate question. There are several ways to go about that that probably depend on how we imagine that happening. For example, if the two temperature sensors are meant to be used in tandem to get a single reliable reading, perhaps the kernel layer should actually combine them. Alternatively, if they are used for totally different purposes (the built-in NRF sensor is mostly for monitoring CPU temperature, while the si7021 for monitoring ambient temerature), those are really different calls.

Ultimately the goal of these user-level interfaces is to be able to write portable code. Inherently, that's going to leave out some use cases, but edge cases should be covered by board-specific system call interfaces and libraries (see #528 for the fuller discussion of this)

niklasad1, 2017-07-28:
Yes, let me know if I should submit PR if it easier to review 😃

alevy, 2017-07-28:
This is a great thread. Let's get the design more or less agree on then do a PR. My guess is it's pretty close.

alevy, 2017-07-28:
A couple comments:

- Do we need periodic functions? In what fraction of the time will those simply be implemented using a virtual alarm in software? If it's most of the time, I think we should just leave that functionality to the client of the temperature HIL. (and if not, they should be called `_periodic`)

- I think it makes sense to separate temperature and humidity. It's only the SI7021 of the three we currently have that measures both, and it's pretty likely for boards to have separate sensors (and even more likely for processes to interact with them separately regardless of whether their implemented on the same chip). So that just means having separate HILs.

- If we're going to combine humidity and temperature, we ought to have a callback for each. In particular, the `measurement_type` argument is just an arbitrary number. How is a client/implementer meant to know which number means what? An enum would be better, or separate callback functions even better, IMO. (or if the HILs for temperature and humidity are separate, maybe that argument goes away completely, which resolves this point as well).

alevy, 2017-07-28:
BTW, I know that last point diverges from how `ninedof` is currently implement. So I'm kind of saying that design is imperfect too.

niklasad1, 2017-07-28:
Good comments, 

> Do we need periodic functions? In what fraction of the time will those simply be implemented using a virtual alarm in software? If it's most of the time, I think we should just leave that functionality to the client of the temperature HIL. (and if not, they should be called _periodic)

Probably not, I think the interface will be more *"clean"* and easier to understand without periodic functions so I agree with you and leave it to the implementer of the temperature HIL. 

> I think it makes sense to separate temperature and humidity. It's only the SI7021 of the three we currently have that measures both, and it's pretty likely for boards to have separate sensors (and [...]

Agree, that is good idea but the actual implementation SI7021 of at moment combines both humidity and temperature thus it need to fixed. I haven't follow your discussion about standardization very closely but perhaps it makes sense to create a HIL named sensors and define these different traits i.e.,
  - TemperatureSensor
- TemperatureClient
- HumiditySensor
- HumidityClient
- etc

> If we're going to combine humidity and temperature, we ought to have a callback for each. In particular, the measurement_type argument is just an arbitrary number. How is a client/implementer meant [...]

This goes away with point 2.

closed 2017-08-15 by niklasad1