tock/tock #462: elf2tbf: Must ensure that the data section is 4 byte aligned.

bradjc, 2017-06-26:
I was testing putting the package name immediately after the TBF header and found that it only works with package names that are multiples of 4 bytes. Somehow this has just been working out for us but we should be more explicit about checking and padding the binary so that the executable code is 4 byte aligned.

Here is what happens if it is on an odd byte (the package name is 5 bytes long):

```
Kernel panic at /home/bradjc/git/tock/kernel/src/process.rs:549:
	""blinkyy" process image invalid. init_fn address must end in 1 to be Thumb, got 0x3007C"
	Kernel version 6a2df4f1
```

Here is what happens if the package name is 2 bytes long:

```
Kernel panic at /home/bradjc/git/tock/kernel/src/process.rs:317:
	"Process blinky had a fault"
	Kernel version 6a2df4f1

---| Fault Status |---
Data Access Violation:              true
Forced Hard Fault:                  true
Faulting Memory Address:            0x66024800
Fault Status Register (CFSR):       0x00000082
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: blinky   -   [Fault]
 Events Queued: 0   Syscall Count: 4   Last Syscall: COMMAND

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ ▼ Grant         272 |   1024          
  0x20005EF0 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ Unused
  0x20004A08 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ ▲ Heap            0 |   1024              S
  0x20004A08 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ R
             │ Data            520 |    520              A
  0x20004800 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ M
             │ ▼ Stack          80 |   2048          
  0x200047B0 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈
             │ Unused
  0x20004000 ┴───────────────────────────────────────────
             .....
  0x00030800 ┬───────────────────────────────────────────
             │ Unused
  0x00030052 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ F
             │ Data            162                       L
  0x00030620 ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ A
             │ Text           1446                       S
  0x0003007A ┼┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈┈ H
             │ Header          116
  0x00030000 ┴───────────────────────────────────────────

  R0 : 0x000000FA    R6 : 0x00000003
  R1 : 0x00000002    R7 : 0x00000000
  R2 : 0x00000002    R8 : 0x00000000
  R3 : 0x46020000    R10: 0x00000000
  R4 : 0x200047DF    R11: 0x00000000
  R5 : 0x00000000    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047B0 (Process Stack Pointer)
  LR : 0x0003009D [0x80000022 in lst file]
  PC : 0x00030282 [0x80000208 in lst file]
 YPC : 0x0003007B [0x8002FAD4 in lst file]

 APSR: N 0 Z 1 C 1 V 0 Q 0
       GE 0 0 0 0
 IPSR: Exception Type - Thread Mode
 EPSR: ICI.IT 0x00
       ThumbBit true
```

ppannuto, 2017-06-26:
Some light reading here, the AAPCS requires that function calls be aligned on 8-byte boundaries: http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka4127.html (albeit only at the boundary of compilation units, so the compiler handles this properly in all cases except the initial frame that we set up)

While I don't _think_ there's any instruction in the Cortex-M family that would require 8-byte (as opposed to 4-byte) alignment, it looks like RIOT actually ran into some issues where clang exploited this and did the wrong thing with their initial stack frame on a 4-byte boundary: https://github.com/RIOT-OS/RIOT/pull/2976

I think when we fix this we should ensure our initial frame is on 8-byte alignment

bradjc, 2017-07-22:
This is taken care of with TBF header v2.

closed 2017-07-22 by bradjc